arduino/ArduinoCore-samd #444: Firmware shorts IO pin for USB OTG devices?

kierenj, 2019-08-19:
https://github.com/arduino/ArduinoCore-samd/blob/020b419fc1dbca49b23b7c0b5e47d50fb2a04485/cores/arduino/USB/samd21_host.c#L145

This code sets PIN_USB_HOST_ENABLE high - but looking at the MKR WiFi 1010 schematic, it's also connected to the USB ID pin. In USB cables, one end of this is tied to GND.

It feels to me like:

1) The board was designed so that PIN_USB_HOST_ENABLE was intended to be an input?
2) Writing it high means that when a cable is inserted, it's grounding an output thats high

Kind of related to #440. I'm just trying to work out what the deal with this pin is. When I try to write it LOW on my board, the IO only goes to 0.3V, not 0V, and the OTG pin on the power IC rises to just 0.1V, when I'm trying to make it go high.

pnndra, 2019-08-20:
Hi,
USB cables meant to connect to devices short the I'd pin to ground however if you use otg cables and connect to the host end it will be left floating hence the pin will go high. The pin connection to the processor is meant as an input so that the processor can sense that the host role is requested and it can then properly set up usb stack

kierenj, 2019-08-21:
I see, thank you. Is there any way I could use this opportunity to understand the official situation with enabling the boost feature on the BQ24195L IC via the OTG pin?

The SAMD core, as linked above, does indeed look to me set the pin you mention to an output, and set it HIGH - so it's not used as an input it seems.

As such, it seemed to me, and the 5 or so random strangers and EE enthusiasts I've desperately accosted across the forum, the internet and beyond, that this I/O pin (PA18), was intended to be used to toggle the OTG pin on the BQ24195L. Is that not the case? I.e., are we not intended to be able to control OTG at all? Is there no way to enable this functionality?

(Would it be possible to explain the samd21_host.c using it as an output - is that a mistake?)

pnndra, 2019-08-23:
Hi,
Setting the pin as an output is wrong and we're fixing this. At most pin shall be set as an input as it would be shorted by the cable in some cases.
Let me also add that we are actually working on releasing a library for the battery charger which will allow setting its registers appropriately for the various conditions. This includes setting charge current limit, otg mode etc.

kierenj, 2019-08-23:
Oh that sounds great!

Although - would you be able to confirm? -

If PA18 is intended as an input, you're intending only for OTG devices connected via USB to trigger "boost mode". As I understand it, without the OTG pin going high, it's not possible to enter "boost mode" and get 5V without a connection, and without a USB device connected requesting it.

So - we won't be able to get to that scenario: have the board generate 5V completely on its own from the battery, no USB device connected?

(In which case, I'll improve my desoldering skills :))

facchinm, 2019-08-23:
I took a closer look and there's a reason for that pin to be set as output (only in Arduino Zero board though).
https://content.arduino.cc/assets/Zero_V1.0_sch.pdf
[image]
[image]
So it may be a good idea to leave the code as is and add a check on the `#ifdef` to make sure we are only applying it on Zero / M0 Pro

pnndra, 2019-08-23:
Hi,
the pin should be set as input in the core as we think the most common application may be to have an USB device. of course in the sketch you are free to change that and set that as an output. the point is that we should prevent hardware issues so if something may break we should avoid it. then if you know what you're doing you can change the direction and force the PMIC to generate the 5V for your shield to operate completely standalone.

kierenj, 2019-08-23:
Ah - so it should work (that would be great!)?

With my first board, I set the IO low, wrote the register, but no luck: measuring the PMIC OTG pin, it was not flipping high. It turned out my FET (Q3) was blown.  I then noticed that R26 was missing, I don't know if that caused Q3 to fail?

I then ordered a new board, it was R26 missing too. I don't want to boot it up for fear that that missing resistor is what caused Q3 to blow.

Could my situation be summarised as - it should be possible for me to set OTG high by setting PA13 low, but it failed because R26 was missing as a manufacturing fault on my board?

kierenj, 2019-08-29:
@pnnda forgive me for the bump here, I really appreciate the attention and help. I have a bit of a deadline - it's few months away but if I need to pivot to another board it'd be great to be able to do so sooner rather than later.

Would it be possible to say - is it possible/supported/supposed to work set set `PIN_USB_HOST_ENABLE` low, and set I2C registers, to enable boost mode (does it work for your boards, for instance)? Was it just R26 stopping me doing so? If so, how can I order boards which definitely have R26 present (and return my two)?

Thank you very much.

pnndra, 2019-08-29:
sorry for the lag. you are going to receive a preview of the library we're developing for the pmic. @Rocketct will take care of it.
let me know if you have issues with it.
thanks,

Dario

Rocketct, 2019-08-29:
Hi @kierenj attach you'll find a zip with the BQ24195 library, two things, inside there are some example that show how to handle basic stuff like enable boost mode and enable charge.
Second we have decide to allow a granular management of the registers bits, then you find a lot of API to enable and disable transparently each PMIC stuff, if you don't find a public one take a look on the protected section in .h.
As we say this is a early preview and may could change in the future.

Riccardo

[Arduino_PMIC.zip](https://github.com/arduino/ArduinoCore-samd/files/3556189/Arduino_PMIC.zip)

kierenj, 2019-08-29:
That's awesome, thank you! I can see however that the boost mode code doesn't do anything about the OTG pin, which is the other requirement for enabling it.

Code aside, I'm really stuck because I still don't know if it's officially supported - to control the BQ24195's OTG pin and generate 5V without any USB device connected. Is it supported - have you had it working on your boards which aren't missing R25 like mine?

Effectively I want to get to a point where that's working - but I don't know why it's not.  I think one of these 3 options needs to be true:

1) It's not supported, we're not supposed to set `PIN_USB_HOST_ENABLE` to `LOW` to set it high (was that ever intended?), it will never work

2) It is supported: setting the I2C reg (directly or via this library), plus `digitalWrite(PIN_USB_HOST_ENABLE, LOW)` will indeed enable boost mode, but it doesn't work for me simply because my boards are missing R25

3) Something else

Are you able to confirm which of these is the situation? I'm really, really hoping it's option 2 - and hoping you can confirm that you have that scenario working?

If that is the case, I just need to know how to return my boards which do not have R25, and where I can buy boards that do work.

(If its instead option 1 or 3, I'd love to know too, since I can then move on and switch to a different product to achieve what I need!)

Thanks again for your advice and patience: I really just want to know if this scenario is supported and how to get there if so. Thank you :)

Rocketct, 2019-08-30:
hi @kierenj, was not the R26? i have make different test, really, to prove that 5V are out from the tested boards, i had fed the board by a 3.7V 2200 mAh battery .
in my test i have:

1) upload the PMICBoostMode sketch;
2) plugged a otg cable;
3) **measured** the voltage between gnd and VBus on the otg cable;
4) connected a second board to the otg cable with the blink example sketch;
5) **measure** the 5V pin voltage (is directly connected to the usb vbus)
6) **measure** the 5V pin voltage (directly on the usb connector of the guest board)

the results are the following:

|  board | cable | a second board connected |
|---|---|---|
| MKR1010(with R25 and r26)   |    5v on cable vbus pin    |     5V over 5V pin of the guest board and 5v pin on usb connector |
| MKR1010 (no R26)       |       3.6V on cable vbus pin     |   3.6V over 5V pin of the guest board 5v pin on usb connector |

Then in the boards where R26 is missing the guest usb receive only 3.6 V from the usb.
No the only board where this library will change the host pin state explicitly will be the Arduino zero is not planned to support the `PIN_USB_HOST_ENABLE` pin control in  Arduino_pmic library, but nothing avoid you to make operation on it you should only set as Input.

in order to prove the correct behavior of the otg i make also another simple test, using the sketch enableBoost and enable charge:

- with enableBoost(that enable the otg mode) when  i connect a smartphone it recognize the host device and start the charge;
- with enableCharge(that disable explicitly the otg mode) when  i connect a smartphone nothing happen.

i make same test with a board where r26 is missing and in that cases the smartphone not respond in both cases and this without any operation on the HOST_PIN.

 i hope this could help you.

Rocketct, 2019-08-30:
NOTE: i get the seame results als witho a normal usb cable

Rocketct, 2019-08-30:
NOTE: you can get 5V also without the cable and by set the PIN_USB_HOST_ENABLE to LOW , but this means that you can't use the usb.
I make also the inverse test if you set the PIN_USB_HOST_ENABLE to LOW in the faulty one,  on the 5V pin you get the 3.6 volt like when the otg cable is plugged to the board.

if you want the 5V without the cable plugged you have to add in the sketch:
```  pinMode(PIN_USB_HOST_ENABLE, OUTPUT);
    digitalWrite(PIN_USB_HOST_ENABLE,LOW);```

you have also to change the boards with a no-faulty one, i'll report to who are in charge to do it :)

kierenj, 2019-08-30:
That's perfect, thank you very much!! I just need some replacement boards then. I'll pick that up on the other github issue where I reported it. Thank you! :)

Rocketct, 2019-08-30:
welcome, i'll let you know when the request will start and you will be contacted by the support

Rocketct, 2019-09-02:
hi @kierenj when you want send the mail to the support for replace specifying for the r26 missing!

kierenj, 2019-09-02:
Thanks, I missed your suggestion to email them I think, I've sent an email now.

kierenj, 2019-09-09:
Thanks - two replacement boards arrived today, and I'm generating 5V on battery by writing one register and setting `PIN_USB_HOST_ENABLE` to `LOW` :)

I'm detecting when USB 5V power is connected/disconnected, but not yet solved getting it charging when reconnected again. I think that one will be easier.

I guess this bug/issue here may still be useful for tracking the fact that the pin is set HIGH, which is a problem for OTG devices? Though I don't care about that scenario - I'm not connecting any :)

kierenj, 2019-09-09:
Ok, sorry, I maybe spoke to soon.

I thought I could monitor `PG_STAT `(bit 2) of `REG08`: because when NOT in OTG mode, this is `1` if the 5V USB power is connected, and `0` if it is not.

However if I enable OTG mode, the BQ24195 of course is not charging, and this `PG_STAT` bit is always `0` (for power not good)... even if the 5V USB power is connected.

I really want to use OTG mode when USB is *not* connected, and non-OTG ("charge") mode when USB *is* connected.

So - is there some other way I can determine if USB is connected, please? Maybe some other pin or mechanism on the microcontroller? Ideally I'd know if any 5V cable is connected, but worst-case I'd be OK to know if an actual USB host is connected and communicating? Thank you again.

engeen-nl, 2020-08-18:
Picked up on this thread because I'm also having trouble generating 5V with the MKR 1310 which also uses the BQ24195L. With the following code I'm managing to make the ON led go on and off when only battery powered. However the voltage on the 5V pin goes from 2 V to approx 2.7 V. I've tried 3 different boards on which R26 is present.

```
#include <BQ24195L.h>
BQ24195L batteryManager;
void setup() {
 pinMode(PIN_USB_HOST_ENABLE, OUTPUT);
 digitalWrite(PIN_USB_HOST_ENABLE, LOW);
 Serial.begin(9600);
 delay(2000);
 Serial.println("BQ24195L starting (delay 3 sec)...");
 delay(3000);

 batteryManager.writeRegister(1, 0b11101011); // Reset all settings to default. Host mode, Charge, Usys_min 3.2 V
 delay(50);
 batteryManager.writeRegister(1, 0b00101011); // Host mode, Charge, Usys_min 3.2 V
 delay(50);
 batteryManager.writeRegister(5, 0b10000110); // Enable termination, match Iterm, disable I2C watchdog timer, disable charging timer

 delay(1000);
}
void loop() {
 batteryManager.writeRegister(1, 0b00011011); // Charge, Usys_min 3.5 V
 delay(3000);

 batteryManager.writeRegister(1, 0b00111011); // Host mode, Usys_min 3.5 V
 delay(3000);
}
```

Writing PIN_USB_HOST_ENABLE HIGH makes the voltage a bit lower and the ON led have reduced brightness. Setting it as an input will disable Host mode altogether.

I also notice it's very important to have a delay after writing to the BQ24195L, otherwise it won't work.

The voltage of the battery I measured was exactly 3.7V during this test. Any ideas why I'm not getting 5V on my 5V pin??

ninouing, 2021-03-03:
> Ok, sorry, I maybe spoke to soon.
> 
> I thought I could monitor `PG_STAT `(bit 2) of `REG08`: because when NOT in OTG mode, this is `1` if the 5V USB power is connected, and `0` if it is not.
> 
> However if I enable OTG mode, the BQ24195 of course is not charging, and this `PG_STAT` bit is always `0` (for power not good)... even if the 5V USB power is connected.
> 
> I really want to use OTG mode when USB is _not_ connected, and non-OTG ("charge") mode when USB _is_ connected.
> 
> So - is there some other way I can determine if USB is connected, please? Maybe some other pin or mechanism on the microcontroller? Ideally I'd know if any 5V cable is connected, but worst-case I'd be OK to know if an actual USB host is connected and communicating? Thank you again.
> 
> (`USBDevice.connected()` doesn't seem to work reliably..)

Hi Kierenj, first of all thank you for all your discussions about enabling the OTG mode, very useful.

One question, where you able at the end to detect when the USB is reconnected again? like you, I would like to enable OTG when USB is disconnected and then disable it when USB is present. 

Thanks in advance!

ninouing, 2021-03-10:
Hi @kierenj, sorry if I insist but this is quite problematic for my current project. Can you let me know if you solved it or not?

Thanks

kierenj, 2021-03-10:
@ninouing sorry, I haven't worked on the project in about a year and don't have time to dig it out now. Hope someone else can help.