arduino/ArduinoCore-samd #446: analogRead() provides incorrect results on the SAMD21 core.

omzlo, 2019-08-28:
Consider the following sketch:

```
#define Debug Serial

void setup() {
  analogReadResolution(12);
  pinMode(A0, INPUT);
  Debug.begin(9600);
  while (!Debug);
}

void loop() {
  int a = analogRead(A0);
  float v = 3.3*((float)a)/4095;

  Debug.print(a);
  Debug.print(" ");
  Debug.print(v,3);
  Debug.println();

  delay(1000);
}
```

If you provide a voltage on pin A0, then results in `v` are typically offset by +35mV (sometimes a little less, sometimes a little more). 

There is likely a bug in the Arduino code of the `analogRead()` function, as detailed here:
https://www.omzlo.com/articles/your-arduino-samd21-adc-is-wrong-did-you-notice

lxrobotics, 2019-09-03:
Hi  @omzlo. Thank you for raising this issue. While you are correct that the ADC error can be improved by waiting for the first ADC run to complete (which is fixed in #447) there is also a mistake within your demo-sketch affecting your accuracy:

The resolution Q of any analog digital converter is calculated by dividing the voltage range by the number of voltage intervals. In case of the SAMD-core with a analog resolution of 12 Bit we do have `2^12 = 4096` voltage intervals -> `Q = 3V3 / 4096` and not `Q = 3V3 / 4095` as you've used in your sketch. 

Depending on how accurate you want your ADC results to be I'd highly recommend to perform offset and gain calibration (against calibrated voltages sources / high resolution voltage measurement devices) as detailed [here](http://ww1.microchip.com/downloads/en/DeviceDoc/90003185A.pdf).

omzlo, 2019-09-03:
Hi @lxrobotics  

Thanks for fixing this bug so quickly. 

You are absolutely correct that the most accurate result will come with a division by 4096 and not by 4095. We should have used that value in our test. The advantage of a division by 4095 is that it will fully scale the results between 0 and 3.3V (instead of 2.9919...), which is why this approach is often used. More importantly here, the division by 4095 introduces an error that is less than 0.9mV here in the absolute worst case, and is negligible towards zero. This has no impact on the issue highlighted here, which creates offsets of about 30mV over all the range of measurements.

In fact, a bigger source of inaccuracy that is only briefly mentioned in the linked blog post is the value of VCC. For simplicity, the sketch assumes that VCC is 3.3V, but in reality in might well be somewhere between 3.28V and 3.32V with typical LDOs we've seen. If VCC is 2.28V, for example, it can introduce a worst case error of 20mV towards 3.3V. Gain calibration can clearly help here. 

Overall, after all these test, we have found that, if used correctly, the ADC of the SAMD21 is quite accurate "out of the box" even without any calibration. It's seems like a good thing that the Arduino community can take advantage of it as well.

PR #447 (2019-09-03): Bugfix - waiting for first ADC read to complete after ADC enable.
This PR fixes #446.
closed 2019-09-03 by lxrobotics