arduino/ArduinoCore-samd #452: Reading Page SIze and Pages does not work

Gadgeteering, 2019-09-16:
Reading the Page Size and Pages on SAMR21x16 is unreliable.

### This Code should be replaced with
uint32_t pageSizes[] = { 8, 16, 32, 64, 128, 256, 512, 1024 };
pageSizes[NVMCTRL->PARAM.bit.PSZ];
NVMCTRL->PARAM.bit.NVMP;
MAX_FLASH = PAGE_SIZE * PAGES;

### by

#Define MAX_FLASH =64 * 1024

sandeepmistry, 2019-09-16:
Hi @Gadgeteering, is this in regard to the boot loader, would you be interested in opening a pull request?

Gadgeteering, 2019-09-24:
I am still investigating this issue along with several other issues. But the data sheet from ATMEL states that it is unreliable.

Gadgeteering, 2020-10-27:
I have found that ATMEL do not provide good support for these devices can you please close this issue

jandrassy, 2020-10-27:
> 
> 
> I have found that ATMEL do not provide good support for these devices can you please close this issue

because they sold it Microchip.

you created the issue, you can close it.

the context is still unclear here. where is this code which should be replaced?

this repository is for boards with SAMD21 

where di you find that it is unreliable?  This " Not all device families will provide all the page sizes indicated in the table."?

matthijskooijman, 2020-10-27:
Closing as requested, and also because it is still unclear where this incorrect code actually is, so there's not much actionable right now.

closed 2020-10-27 by matthijskooijman