arduino/ArduinoCore-samd #457: New to SAMD21 - is there no Serial.printf?

beegee-tokyo, 2019-09-30:
Maybe it's just me, but on other architectures I am have no problems using **_printf_** like
```CPP
Serial.printf("\nLoRa begin failed %d\n", state);
```
But for SAMD21 it throws compile error
```LOG
src\main.cpp:78:10: error: 'class Uart' has no member named 'printf'; did you mean 'print'?
   Serial.printf("\nLoRa begin failed %d\n\n", state);
 ```

Am I missing something

matthijskooijman, 2019-09-30:
No official architecture supports `Serial.printf` AFAIK (intentionally, there are some discussions on the mailing list and github about this). There might be third-party libraries that make it work, maybe you have some such library installed for other archs? Or maybe you are comparing with third-party architectures?

beegee-tokyo, 2019-09-30:
With other architectures I meant ESP32 Arduino, ESP8266 Arduino, nRF52 Arduino frameworks.
Also printf is standard C++ http://www.cplusplus.com/reference/cstdio/printf/

Surprised it is not on SAMD

beegee-tokyo, 2019-09-30:
Example. on ESP32 it is in Print.cpp
```cpp
size_t Print::printf(const char * format, ...)
{
  char buf[256];
  int len;

  va_list ap;
  va_start(ap, format);

  len = vsnprintf(buf, 256, format, ap);
  this->write(buf, len);

  va_end(ap);
  return len;
}```

Wondering if that would work on SAMD

beegee-tokyo, 2019-09-30:
I see, that it is not in ArduinoAVR framework. Maybe I was just lucky with the platforms I used before.

closed 2019-09-30 by beegee-tokyo

matthijskooijman, 2019-09-30:
Yeah, those are third-party platforms that have made this custom addition. It is not part of the official Arduino API, for reasons previously stated elsewhere. See also https://github.com/arduino/ArduinoCore-API/issues/34.