arduino/ArduinoCore-samd #463: micros() is non-monotonic with -1ms glitches on Cortex cores from priority 0 or 1 ISRs

colintd, 2019-10-16:
Whilst using micros() to timestamp hardware events on a SAMD21 processor, I've discovered that the function is broken if called from priority 0 or 1 ISRs.  Specifically, from time to time, the result is 1ms behind, making the value returned non-monotonic and inaccurate.

Looking at the code, the micros() function has logic to determine if there is a pending SysTick ISR, and adds 1ms() if the SysTick counter has rolled, but the tick increment ISR has not been called.  This works fine when the SysTick ISR has priority 0, as it is either pended (if another priority 0 ISR is running already), or runs to completion.

However, pull request 182, related to adding I2S support, reduced the SysTick ISR to priority 2 (from priority 0).  

```
/cores/arduino/wiring.c

NVIC_SetPriority (SysTick_IRQn,  (1 << __NVIC_PRIO_BITS) - 2);  /* set Priority for Systick Interrupt (2nd lowest) */
```

This means that a priority 0 or 1 ISR (such as that generated by a GPIO interrupt) can preempt an already running SysTick ISR.  Under these conditions the logic to add the required 1ms never activates (as the ISR is _active_ not _pending_ ).
 
If the SysTick ISR has already incremented the tick count, all is fine.  If it hasn't, the result of micros() effectively loses 1ms, which makes the result inaccurate and non-monotonic.  (Note, even if you checked whether the SysTick ISR is active, it isn't possible to determine whether the tick count has or hasn't been incremented from within micros())

I'm hitting the problem timestamping precision edges, and see the issue whenever an edge aligns with a roll of the SysTick.  With 1PulsePerSecond edges I get the incorrect result a few times an hour, as the wander of the main clock leads to the SysTick roll aligning with the external edge.  

Alternatively, I suspect that if you use a non-precision 1KHz+ input, you will see frequent 1ms backsteps.

I'm not aware of why it was felt necessary to reduce the priority of SysTick to get I2S to work, but I can confirm that restoring SysTick to priority 0 returns micros() to being monotonic and accurate from ISRs of priorities 0 & 1. I'd suggest that given the critical nature of the micros() function, SysTick should be restored to priority 0, and an alternative solution found for the I2S problem(s).

sslupsky, 2019-10-23:
@colintd Looking at the commit history for that change it may have something to do with the DMA.

I do not think the SysTick interrupt is broken.  Rather, micros() appears to have a bug where it introduces jitter if an interrupt occurs during the call to micros().  I had a look at micros() and it seems to me the correct way to calculate micros() would be to guard the SysTick register and ms counter to ensure coherency during read so that an interrupt cannot cause jitter?  

For example:

```
unsigned long micros( void )
{
  uint32_t ticks;
  uint32_t count;

  noInterrupts();
  ticks  = VARIANT_MCK / 1000 - SysTick->VAL;
  count  = _ulTickCount;
  interrupts();

  return (count * 1000) + ((ticks*(1048576/(VARIANT_MCK/1000000)))>>20) ;
  // this is an optimization to turn a runtime division into two compile-time divisions and
  // a runtime multiplication and shift, saving a few cycles
}
```

sslupsky, 2019-10-23:
The comment at the beginning of the micros() function indicating it is "interrupt compatible" may be misleading.  I think someone was thinking micros() could be crafted so that if an interrupt occurs, then they detect it and correct it.  This is quite simply impossible because the non atomic register access imposes race conditions.  No amount of "looping" around will guarantee that you avoid the race ... I think this is what you have bumped into.

I looked at the commit history and unfortunately, this version of micros() was in the first commit so there is no insight to be gained regarding the history of this function.

sslupsky, 2019-10-23:
I updated my code snippet to ensure that "ticks" and "count" are treated as volatile.

There is still one potential race condition though.  If the SysTick interrupt is being processed and a higher priority interrupt occurs, if the ISR of the higher priority interrupt calls micros(), then it is possible that the SysTick counter rolled over but the ms tick count has not yet incremented.  In this case, the time would appear to briefly jump back by up to 1ms.  The interrupt pending indication will only tell you that it was in the SysTick ISR but you do not know where so you cannot know if the ms counter has been incremented yet.  There might be a way to compensate if you examined the program counter on the stack you know where it was interrupted.  Otherwise, it is probably best to avoid using micros() in an ISR?

colintd, 2019-10-24:
Btw, with the SysTick restored to the original priority, there is absolutely no need to have noInterrupt and Interrupt round the access in micros().  As written it is 100% correct, and works perfectly.

sslupsky, 2019-10-24:
I tried to figure out what was "fixed" but it is not clear ... its part of some DMA work that was done so that is why I presume it had something to do with the DMA.  My "guess" is that the tick was causing hiccups with the DMA transfers but I have no certainty that is the case.  @sandeepmistry might be able to shed some light on this.

I do not disagree with increasing the priority level of the SysTick interrupt as it does solve the issue.  The SAMD SysTick was poorly designed at the hardware level and there are a few shortcomings.  This issue is one of them.  Another issue is that SysTick is shut off during sleep which breaks micros() and millis().

If you restored the priority level, which is more efficient?  All that pending interrupt looping stuff with additional memory or the atomic read?

colintd, 2019-10-24:
I’d very strongly go for the existing code structure, for several reasons:

  *   you only hit the loop under a very small number of edge conditions when the SysTick interrupts a lower priority caller
  *   by avoiding the noInterrupt/interrupts calls you avoid normal priority code disturbing the timing of ISR
  *   you get cleaner behaviour with nested layers of interrupts
  *   most importantly, it was well tested code with no visible problems until the I2S change, so I rather leave as it.

Does that make sense?

colintd, 2019-10-24:
Excellent.

As I said I've been testing for the last couple of weeks, including with high frequency gpio isrs.  With the low priority systick I'd hit the problem reliably in under a minute.  Restoring to priority 0 I've never see the -1ms jump, and I'm getting ISR timing with RMS jitter under 15us from a precision 1pps source.

The pain is that the low priority systick has now been copied into lots of variant files, but I guess we've got to start somewhere with the fixing.

Do you want to apply the change, or do you need a pull request from me?

Cheers,

Colin.

I think so.

colintd, 2019-10-25:
Actually in latest testing (a few tweaks to my code), I'm down to ~5us RMS jitter using the micros() code with restored priority..  This is perfectly acceptable performance for most uses.

sslupsky, 2019-10-25:
@colintd I am not the maintainer so I cannot merge anything into this repo. Likely either @sandeepmistry or @facchinm will review this but it may take a while.  The more information posted here about this the more likely they will ask for a PR.  I think it would be a good idea to submit the PR referencing this issue.

matthijskooijman, 2019-10-28:
I just read up on this issue and I agree with the analysis (though I haven't looked at the code myself). I also think the proposed solution (restore the systick priority to 0 and fix I2S in another way if possible). I also think the current code with pending interrupt looping should be preserved, since I do not think that the atomic `micros()` helps in all cases (it won't help when `micros()` is called from an interrupt that preempted the systick ISR AFAICS).

I'm no maintainer either, but I 2wo

> The pain is that the low priority systick has now been copied into lots of variant files, but I guess we've got to start somewhere with the fixing.

Really? Isn't the only changed needed to remove https://github.com/sandeepmistry/ArduinoCore-samd/blob/284bc5a3185756cf6d107c002cc8841d772d07d4/cores/arduino/wiring.c#L66 ?

colintd, 2019-10-28:
@matthijskooijman I agree wiring.c is the root change, but doing a quick grep of the installed code on my system for NVIC_SetPriority then SysTick_IRQn I can see this change has propagates in lots of SysTick_Config functions, bootloader and the adafruit cores files (see below).

packages/arduino/hardware/samd/1.8.3/bootloaders/sofia/Bootloader_D21_Sofia_V2.1/src/ASF/thirdparty/CMSIS/Include/core_cm0plus.h:  NVIC_SetPriority (SysTick_IRQn, (1<<__NVIC_PRIO_BITS) - 1);  /* set Priority for Systick Interrupt */
packages/arduino/hardware/samd/1.8.3/bootloaders/mzero/Bootloader_D21/src/ASF/thirdparty/CMSIS/Include/core_cm0plus.h:  NVIC_SetPriority (SysTick_IRQn, (1<<__NVIC_PRIO_BITS) - 1);  /* set Priority for Systick Interrupt */
packages/arduino/hardware/samd/1.8.3/cores/arduino/wiring.c:  NVIC_SetPriority (SysTick_IRQn,  (1 << __NVIC_PRIO_BITS) - 2);  /* set Priority for Systick Interrupt (2nd lowest) */
packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/core_cm4.h:  NVIC_SetPriority (SysTick_IRQn, (1UL << __NVIC_PRIO_BITS) - 1UL); /* set Priority for Systick Interrupt */
packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/core_cm3.h:  NVIC_SetPriority (SysTick_IRQn, (1UL << __NVIC_PRIO_BITS) - 1UL); /* set Priority for Systick Interrupt */
packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/core_sc000.h:  NVIC_SetPriority (SysTick_IRQn, (1UL << __NVIC_PRIO_BITS) - 1UL); /* set Priority for Systick Interrupt */
packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/core_cm0plus.h:  NVIC_SetPriority (SysTick_IRQn, (1UL << __NVIC_PRIO_BITS) - 1UL); /* set Priority for Systick Interrupt */
packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/core_sc300.h:  NVIC_SetPriority (SysTick_IRQn, (1UL << __NVIC_PRIO_BITS) - 1UL); /* set Priority for Systick Interrupt */
packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/core_cm7.h:  NVIC_SetPriority (SysTick_IRQn, (1UL << __NVIC_PRIO_BITS) - 1UL); /* set Priority for Systick Interrupt */
packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/core_cm0.h:  NVIC_SetPriority (SysTick_IRQn, (1UL << __NVIC_PRIO_BITS) - 1UL); /* set Priority for Systick Interrupt */
packages/adafruit/hardware/samd/1.5.4/bootloaders/sofia/Bootloader_D21_Sofia_V2.1/src/ASF/thirdparty/CMSIS/Include/core_cm0plus.h:  NVIC_SetPriority (SysTick_IRQn, (1<<__NVIC_PRIO_BITS) - 1);  /* set Priority for Systick Interrupt */
packages/adafruit/hardware/samd/1.5.4/bootloaders/mzero/Bootloader_D21/src/ASF/thirdparty/CMSIS/Include/core_cm0plus.h:  NVIC_SetPriority (SysTick_IRQn, (1<<__NVIC_PRIO_BITS) - 1);  /* set Priority for Systick Interrupt */
packages/adafruit/hardware/samd/1.5.4/cores/arduino/wiring.c:  // NVIC_SetPriority (SysTick_IRQn,  (1 << __NVIC_PRIO_BITS) - 2);  /* set Priority for Systick Interrupt (2nd lowest) */

Even in the main ArduinoCore-samd.git repo, I can see 3 places the priority is set:

bootloaders/sofia/Bootloader_D21_Sofia_V2.1/src/ASF/thirdparty/CMSIS/Include/core_cm0plus.h:  NVIC_SetPriority (SysTick_IRQn, (1<<__NVIC_PRIO_BITS) - 1);  /* set Priority for Systick Interrupt */
bootloaders/mzero/Bootloader_D21/src/ASF/thirdparty/CMSIS/Include/core_cm0plus.h:  NVIC_SetPriority (SysTick_IRQn, (1<<__NVIC_PRIO_BITS) - 1);  /* set Priority for Systick Interrupt */
cores/arduino/wiring.c:  NVIC_SetPriority (SysTick_IRQn,  (1 << __NVIC_PRIO_BITS) - 2);  /* set Priority for Systick Interrupt (2nd lowest) */

I'm very tempted to add a 

#define SYSTICK_DEFAULT_PRIORITY 0

 in one of the top headers, and then change all 3 references to use that value.  That will then provide a consistent basis for others to bring their variants into line.

Seems reasonable?

sslupsky, 2019-10-28:
Can you tell when the changes to CMSIS occurred?  That is troubling.  

Perhaps @sandeepmistry made the original changes simply to bring the priority level inline with the CMSIS library?

sslupsky, 2019-10-28:
I cannot think of a reason that this change would have an adverse affect on the bootloaders.  It would probably be a good idea to ask @ladyada from Adafruit to loop someone into this discussion.

colintd, 2019-10-28:
These files never seem to have changed, but the initial commits were after the I2S tweak, so I assume they just copied the “default” from there.

matthijskooijman, 2019-10-28:
I can imagine that all these CMSIS mentions are in code that is not actually used. Perhaps you could check that?

As for the bootloader / CMSIS code, as long as you never use the timer from an interrupt and only read the timer values atomically, I guess having a low priority for the systick ISR is not a problem, so maybe that's why that's the default there?

colintd, 2019-10-28:
From the file timings, I suspect these values are just copies of what was in wiring.c at the point they were created, rather than a deliberate decision to use a lower priority.

Similarly, from the history I’m fairly sure the adafruit value is just a merge of the mainline wiring.c change.

My instinct is therefore to change all to use the value of 0, which was the original in wiring.c for a long time.

sslupsky, 2019-10-28:
> As for the bootloader / CMSIS code, as long as you never use the timer from an interrupt and only read the timer values atomically, I guess having a low priority for the systick ISR is not a problem, so maybe that's why that's the default there?

Yes, I think this is probably the underlying issue for the CMSIS people but this is just speculation on my part.  Generally speaking, and in my opinion, you would want to use as low a priority as possible simply because in the general case you may not want the Systick to interrupt something else that has a higher timing requirement.

sslupsky, 2019-10-28:
Those CMSIS includes are used ... I know for certain the core_cm0plus.h is included by way of Arduino.h.  I do not think we (Arduino that is) can just change these files since these are part of the CMSIS library.  The maintainer for CMSIS is ARM?

colintd, 2019-10-28:
In many ways I’m relaxed about the bootloaders because once the program starts the value is overwritten.  I am, however, keen to get the change merged forward into the adafruit code (which was where I actually hit the problem).  If we update wiring.c, then I can followup.

matthijskooijman, 2019-11-05:
> Those CMSIS includes are used ... I know for certain the core_cm0plus.h is included by way of Arduino.h.
Just including does not mean the code actually *runs*. It might very well be in a function that is not called.

> I do not think we (Arduino that is) can just change these files since these are part of the CMSIS library. The maintainer for CMSIS is ARM?

I think so too, I think the CMSIS is a utility library that is intended for ARM cores in general and contains a lot of code that is probably not actually called by Arduino. Given that this issue did not exist before the I2S change (AFAIU), I suspect that the default systick priority is the highest, and the code from CMSIS never runs (and thus also does not need to be modified).

> Generally speaking, and in my opinion, you would want to use as low a priority as possible simply because in the general case you may not want the Systick to interrupt something else that has a higher timing requirement.

That is actually a good point. Can we really not find any way to make things work with a low priority systick?

colintd, 2019-11-05:
I’d actually argue the opposite way in terms of priority.  In my experience systick/timer ISRs almost always run at highest priority to prevent exactly this kind of problem of premption (where you can’t know if the wrap count has incremented) , and to ensure that time “advances” even when a lower priority ISR is running for an extended period.  The tick ISRs are typically very short, so have minimal impact on other functions.  I’ve worked on a number of systems where the tick timer was actually tied to the NMI vector.

PaulStoffregen, 2019-11-05:
> In my experience systick/timer ISRs almost always run at highest priority to prevent exactly this kind of problem

This is how we do things on Teensy.  It has worked out very well.

baekgaard, 2020-07-28:
Sorry to revive an old issue, but I was just being hit by the same problem here on a CM4 (samd51j19) device: 

As the systick handler runs at a lower priority (prio level 5, it seems, in the default adafruit setup (or 7 when using a FreeRTOS layer on top in my case)) I also got a non-monotic clock in my ISR dealing with hardware interrupts; the issue is clearly that my ISR could interrupt a running systick handler when the interrupt was no longer pending but the routine still was in the process of updating the tick count.

The remedy, which was easy enough in my case when I realised the problem, was to lower the priority of my ISR to the same or a lower level than the systick handler. I only found this thread later... I wished I had seen it before :-)

Was this ever resolved by running the systick handler at the highest priority in the standard setup?

If not, perhaps at least a comment in the micros() function could be in place, just to save some time for those that may also run into this issue with non-monotonic clocks.

The micros() code is correct as it is now, but the comment about this being interrupt friendly means that it will return the correct value even when being interrupted in the middle of the execution by the systick interrupt -- but it is NOT currently guaranteed to give monotonically increasing values if called from an ISR that has a higher priority than the systick handler (in fact, it is almost guaranteed to have a -1ms error very frequently (in my case 1-5 times pr. second)).

colintd, 2020-07-28:
I don't think there was yet consensus about the change (due to possible impact on the I2S code), but as noted in the comments by multiple people, systick is normally a top priority interrupt, and as it stands micros() is non-monotonic.  

Since raising the bug I have also been running with the high priority systick without any issues (and with monotonic time).

So my preference would definitely be to merge in the change...

mzero, 2020-10-28:
Sadly, I think there is another problem with `micros()`:

The implementation ends with:
```C++
return ((count+pend) * 1000) + (((SysTick->LOAD  - ticks)*(1048576/(VARIANT_MCK/1000000)))>>20) ;
```

The idea is `count + pend` gives the correct number of milliseconds, whether the interrupt has occured (and just bumped `count`) or not (and so `pend` is 1).  Then the number of microseconds in the counter is added in with `SysTick->LOAD - ticks`.

However, the issue is that the interrupt occurs when the SysTick counter transitions to 0, not when it reloads (on the next clock). 

SO- Consider when `millis()` is called at a moment when the SysTick counter is 0.  Either `count` will have just been bumped or there'll be a pending interrupt.  Either way, `count + pend` is the same.  Now, `(SysTick->LOAD - 0)` will add in almost a full milliseconds worth of microseconds -- even though that millisecond has already been accounted for in `count + pend`.

As a thought experiment, consider if `millis()` is called one clock later - `count + pend` will be the same, but `ticks` will be the re-load value... and so `(SysTick->LOAD - SysTick->LOAD)` will add zero microseconds in... The value will have gone backwards.

rohoog, 2021-04-24:
I can confirm that using `NVIC_SetPriority(EIC_IRQn, 2)` after all `attachInterrupt()` calls solves the 1ms glitches.

learnfromgirls, 2022-04-05:
Thank you all for suggesting the fix of lowering the priority of hardware interrupts and especially @rohoog
I implemented 
NVIC_SetPriority(EIC_IRQn, 2)
after my attachInterrupts and it worked to fix the +2msec and -1msec jitter on timing mains zerocrossings detection.

ProfFan, 2022-12-02:
Even after setting the priority there are still glitches, just less frequent...

rohoog, 2022-12-03:
@ProfFan , of course the priority solution depends on whether you might have other interrupt sources that migh need their priorities tuned.