arduino/ArduinoCore-samd #465: ADD NANO 33 IoT Support to SDU

agdl, 2019-10-24:
Otherwise you get unsupported board https://github.com/arduino/ArduinoCore-samd/blob/master/libraries/SDU/src/SDU.cpp#L44

kaiiorg, 2020-07-30:
Let me preface this with I have no idea what I'm actually doing, I'm just reading the code that I've been able to find and seeing what happens from there.

I ran into this issue today and since there hasn't been any movement on this, I thought I'd look and see if I could figure it out. I got a lot farther than I expected, but this is what I have worked out on my own so far:
1. `SDU.cpp` includes a header file for each type of board, which is where we get the error that the board we want isn't supported
2. In the `extras/SDUBoot/`, it looks like there is a script to compile said header files for each board type
3. At the bottom of `extras/SDUBoot/build.sh` I added `buildSDUBootSketch "arduino:samd:nano_33_iot" "$OUTPUT_PATH/nano33iot.h"`
4. After installing the FlashStorage library and running the script (on a raspberry pi 4 in my case), it appears that the headers are built for each board type
    - I was getting a lot of java warnings about DNSStateTask and DNSResolverTask failing because a requested address could not be assigned. I'm not sure if that's related to my environment or if there's something else going on, but it still built the headers.
5. I then added a case to allow support, but used a different path while I was testing this in my existing project.
```c++
#elif defined(ARDUINO_SAMD_NANO_33_IOT)
  #include "nano33iot.h"
```
6. After adding the modified library to my project, the Verify/Compile fails:
```
SDU.cpp:48:1: error: too many initializers for 'unsigned char [16384]'
 };
 ^
```

On closer investigation, it looks like it is failing because the compiled output is larger than the array I'm trying to squeeze it into; trying to fit some 16,847 elements into a 16,384 long array. I'm fairly sure that _bad things™_ would happen if I try to make that array larger; I'm playing with fire already, given that I honestly don't know what I'm doing here. 

Does anyone have any suggestions on how to trim down the size of the compiled output? It looks like the files resulted from the modified script are significantly larger than the existing files. Or better yet, how to do this correctly : )

winner10920, 2021-08-09:
I believe I've found a solution that works with the IDE 1.8.13 and samd core 1.8.11, I haven't tested it with an sd card yet but it runs and after various attempts and having to force the board back into bootloader it seems to run fine to a test sketch.
Need to modify SDU.cpp in the samd core folder here
Here's my folder as an example 
/home/pi/Desktop/pi/.arduino15/packages/arduino/hardware/samd/1.8.11/libraries/SDU/src/boot

From 
unsigned char sduBoot[0x4000] = {
To
unsigned char sduBoot[0x6000] = {

And add this towards the end before the last else statement
  #elif defined(ARDUINO_SAMD_NANO_33_IOT)
  #include "boot/nano_33_iot.h"

And then in the IDE export compiled binary  this program for your board, in my case the nano 33 iot

https://github.com/arduino/ArduinoCore-samd/blob/master/libraries/SDU/extras/SDUBoot/SDUBoot.ino

After that you need to run the following in the SDU sketch folder, and past the new file into the boot folder of the samd core for example mine is here

/home/pi/Desktop/pi/.arduino15/packages/arduino/hardware/samd/1.8.11/libraries/SDU/src/boot

(I renamed the bin without bootloader to make this line easier during testing"

cat "SDU.bin" | xxd -i > nano_33_iot.h

Then when you goto your sketch all you need to do is the same include sdu and it seems to be working

If anyone is having this same issue I can do a pull request but otherwise not

JEG8, 2022-03-22:
Has there been any resolution to this issue? I have tried what you advised above, but the Arduino hangs during bootloader phase.

winner10920, 2022-03-22:
> Has there been any resolution to this issue? I have tried what you advised above, but the Arduino hangs during bootloader phase.

I haven't been following the past few months but my boards using modifications I posted about above have been solidly working for a long while now, I'll link to my version of the repo [here](https://github.com/winner10920/ArduinoCore-samd/tree/master/libraries%2FSDU)
You should be able to pretty much replace this folder in your samd core libraries folder  to be able to make it work but I haven't tested. Also not sure if it makes a difference but I'm using Linux

winner10920, 2022-03-22:
> Thank you for the quick response! I added a bash shell to windows so I was able to use your commands from ubuntu. So far, the program downloads the file and stores it to the SD, but the board never boots with the UPDATE.BIN... it just hangs after the NVIC reset. I think I have an issue with the bootloader. Am I supposed to be using the binary file with or without the bootloader?
>
> I'll take a look at your repo and let you know if I am able to resolve.

I believe it's without the bootloader, and also make sure the program you are using to download to SD has the SDU library included and then every subsequent binary needs to have that library included in the program so it can continue to have that update from SD feature

winner10920, 2022-03-22:
No problem 👍 glad you got it working

Thank you! I was able to get it to work. It was the one without the bootloader. The issue was that the file on my server did not have the SDU.h library included. Once I added the file, it started right up.