arduino/ArduinoCore-samd #468: arm-none-eabi-gcc / 7-2017q4 precompiled binaries out of date

ardiri, 2019-11-18:
in the current 'package_index.json' the gcc/tools download is defined as:

```
          "name": "arm-none-eabi-gcc",
          "version": "7-2017q4",
          "systems": [
            {
              "host": "arm-linux-gnueabihf",
              "url": "http://downloads.arduino.cc/tools/gcc-arm-none-eabi-4.8.3-2014q1-arm.tar.bz2",
              "archiveFileName": "gcc-arm-none-eabi-4.8.3-2014q1-arm.tar.bz2",
              "checksum": "SHA-256:ebe96b34c4f434667cab0187b881ed585e7c7eb990fe6
b69be3c81ec7e11e845",
              "size": "44423906"
            },
```

however this points to an OLDER (4.8.3 2014q1) version of the compiler for arm-linux-gnueabihf, instead of the required version (7.2.1 2017q4). the windows/macosx/linux variables are correct; however the arm variants have not been precompiled with the right versions and are out of date.

this causes some samd projects to fail to compile when using the tools on a raspberry pi target.

ardiri, 2019-11-20:
for those who come across this - there is no archive provided by arduino.

your only resort is to cross-compile gcc for the gcc-arm-none-eabi target specifically for the raspberry pi. you can download the source archive here where you can must have a spare raspberry pi with the gcc development environment installed. it requires a lot of patience and sufficient file space to complete the compile (15GB free space, 1GB swap required).

https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

i just completed it on my own raspberry pi (ran into a lot of issues) - here are the steps i went through. there was an issue compiling gdb, but i simply skipped over it as we do not require it for the arduino development.

```
>> install gcc dependencies

$ sudo apt-get update
$ sudo apt-get install m4 bison flex texinfo texlive

>> increase swap file to 1GB

$ sudo dphys-swapfile swapoff
$ sudo vi /etc/dphys-swapfile           --> CONF_SWAPSIZE=1024
$ sudo dphys-swapfile setup
$ sudo dphys-swapfile swapon

>> compile

$ ./install-sources.sh --skip_steps=mingw32
$ ./build-prerequisites.sh --skip_steps=mingw32
$ ./build-toolchain.sh --skip_steps=mingw32 --build_type=native
```

good luck :) thankfully i have 7-2017q4 pre-compiled binaries on my system now.

facchinm, 2019-11-20:
We are going to merge the 7-2017q4 for **ARMv7** (so no support for Rpi1 and Zero) by the end of this week. The missing compatibility with older rpi is why we have been delaying the release but it looks like there's no way to actually cross-compile for armv6.
Here's the link to the toolchain http://downloads.arduino.cc/tools/gcc-arm-none-eabi-7-2019-q4-major-linuxarm.tar.bz2

ardiri, 2019-11-20:
i spent two days getting it compiled on my RPi3+ - which, eventually worked 

i had a number of failures; but the most obvious one was lack of RAM/memory - which i was able to resolve by increasing the swap space to 1GB (it was originally 100Mb).  the max swap space size is 2GB, maybe that would be the reason why you couldn't cross-compile for armv6? 

also; shouldn't your file name be renamed correctly?

gcc-arm-none-eabi-7-2017-q4-major-linuxarm.tar.bz2

would be more appropriate for the distribution, since it isn't technically the 2019 version

facchinm, 2019-11-20:
The name is correct since the build script picks the compile time (I used 7-2017-q4-major sources obviously). Native compilation failure on rpi1 is indeed due to ram; a canadian cross compiler would solve the issue but the build script should we reworked heavily to do it

JosefWN, 2024-01-14:
How is this compiler different from the one managed by the ARM community?
https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

It seems the compiler is now at:
```json
  {
    "packager": "arduino",
    "name": "arm-none-eabi-gcc",
    "version": "7-2017q4"
  },
```

Is there a reason you are packing it yourselves? If it's an unmodified copy of the ARM GNU Toolchain I would wager to guess it's discontinued by now, the thing is antique.