arduino/ArduinoCore-samd #470: Please add board SAMD21 M0-Mini

lnlp, 2019-11-24:
I am using a SAMD21 M0-Mini board.
(example: https://robotdyn.com/samd21-m0-mini-not-soldered.html)

The board is available from different manufacturers and labeled either as 'Wemos' or 'Robotdyn'.

I have seen blogs stating that one should select board "Arduino Zero (USB Native Port) for this board. This works but gives problems for SPI.

The problem is that the 'Arduino Zero (USB Native Port)' board definition defines SPI pins as part of the ICSP connector which is on the top of the board but these pins are not available on the side/bottom of the board which is not breadboard-friendly.

Instead I would like to use other pins on the side/bottom of the board for the standard SPI port. This currently appears not to be supported.
Adding a separate board definition for the SAMD21 M0-Mini (where the standard SPI port is defined using pins on the side/bottom) could fix this.

I know that it is possible to define a second SPI port which is described in the following Adafruit article: https://learn.adafruit.com/using-atsamd21-sercom-to-add-more-spi-i2c-serial-ports/creating-a-new-spi

But this does define a second SPI port however and does not redefine the pins for the _standard SPI port_. It is important to be able to use the _standard SPI_ port and not a second SPI port because using the second SPI port would require current code and libraries to be modified.

ardiri, 2019-11-25:
why not copy the variants/mkrzero directory and make a new one - editing the PINS for SPI in the configuration of the variants.h file?

lnlp, 2019-11-25:
I thought about that but that will get overwritten with every update of the SAMD Arduino Core and will require some additional tweaking when using PlatformIO. It much nicer to have a board definition added for this board. I realize that this board is not produced by the Arduino organisation but I would like them  to add its board definition to the standard SAMD Arduino Core. (Similar how Espressif adds all ESP32 board definitions to the ESP32 Arduino Core, also for boards not produced by Espressif).

ardiri, 2019-11-25:
yeah; you could do it. then do a pull request to get it integrated into the official version.

lnlp, 2022-03-21:
> yeah; you could do it. then do a pull request to get it integrated into the official version.

I only have one of these boards for testing to support it for the cross-platform application LMIC-node.
I therefore have no intention of adding support for the SAMD21 M0-Mini myself.