arduino/ArduinoCore-samd #479: SDU.h failing to compile with adjusted boot file (SS pin change from 4 to 6)

salvq, 2019-12-16:
Hello,

Reporting an issue compiling sketch using SDU library with adjusted boot h file. I am using different SS pin 6 instead of 4 therefore I am in need of recompile.

I have MKR GSM 1400 and when compiled example `Usage.ino` using adjusted boot file mkrgsm1400.h compilation is failing on below. I am using Arduino: 1.8.10 (Windows 10), Board: "Arduino MKR GSM 1400"

```
Arduino: 1.8.10 (Windows 10), Board: "Arduino MKR GSM 1400"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\home\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\home\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\home\Documents\Arduino\libraries -fqbn=arduino:samd:mkrgsm1400 -vid-pid=0X2341_0X8052 -ide-version=10810 -build-path C:\Users\home\AppData\Local\Temp\arduino_build_776342 -warnings=none -build-cache C:\Users\home\AppData\Local\Temp\arduino_cache_151218 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.bossac.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.openocd.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.CMSIS.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -verbose C:\Users\home\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.4\libraries\SDU\examples\Usage\Usage.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\home\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\home\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\home\Documents\Arduino\libraries -fqbn=arduino:samd:mkrgsm1400 -vid-pid=0X2341_0X8052 -ide-version=10810 -build-path C:\Users\home\AppData\Local\Temp\arduino_build_776342 -warnings=none -build-cache C:\Users\home\AppData\Local\Temp\arduino_cache_151218 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.bossac.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.openocd.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.CMSIS.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -verbose C:\Users\home\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.4\libraries\SDU\examples\Usage\Usage.ino
Using board 'mkrgsm1400' from platform in folder: C:\Users\home\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.4
Using core 'arduino' from platform in folder: C:\Users\home\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.4
Detecting libraries used...
"C:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10810 -DARDUINO_SAMD_MKRGSM1400 -DARDUINO_ARCH_SAMD -DUSE_ARDUINO_MKR_PIN_LAYOUT -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8052 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino MKR GSM 1400\"" -DUSE_BQ24195L_PMIC "-IC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.4\\cores\\arduino" "-IC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.4\\variants\\mkrgsm1400" "C:\\Users\\home\\AppData\\Local\\Temp\\arduino_build_776342\\sketch\\Usage.ino.cpp" -o nul
Alternatives for SDU.h: [SDU@1.0.0]
ResolveLibrary(SDU.h)
  -> candidates: [SDU@1.0.0]
"C:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10810 -DARDUINO_SAMD_MKRGSM1400 -DARDUINO_ARCH_SAMD -DUSE_ARDUINO_MKR_PIN_LAYOUT -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8052 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino MKR GSM 1400\"" -DUSE_BQ24195L_PMIC "-IC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.4\\cores\\arduino" "-IC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.4\\variants\\mkrgsm1400" "-IC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.4\\libraries\\SDU\\src" "C:\\Users\\home\\AppData\\Local\\Temp\\arduino_build_776342\\sketch\\Usage.ino.cpp" -o nul
"C:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10810 -DARDUINO_SAMD_MKRGSM1400 -DARDUINO_ARCH_SAMD -DUSE_ARDUINO_MKR_PIN_LAYOUT -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8052 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino MKR GSM 1400\"" -DUSE_BQ24195L_PMIC "-IC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\ardui [... cut ...]
The only change I made to SDUBoot.ino is below listed below. I have changed SS PIN and compiled via Arduino IDE 1.8.10 (export export compiled binary) and then use cat in raspberry pi (buster) `SDUBoot.ino.mkrgsm1400.bin | xxd -i > /tmp/mkrgsm1400.h` to create adjusted boot version and rewrite original file in boot folder of library folder.

```
#ifndef SDCARD_SS_PIN
#define SDCARD_SS_PIN 4
#endif
```

**Does anybody knows why the compiler is failing when I changed just the SS pin ? What am I doing wrong ? Is there any settings I need to make in SDUBoot.ino ?**

Thanks for help

facchinm, 2019-12-16:
Hi @salvq ,
this is probably due to the new compiler (arm-none-eabi-gcc/7-2017q4) blowing code size just enough not to fit the resulting binary in 16KB.
You can try reverting the SAMD core to pre 1.8 and rerun your procedure because I don't see anything wrong in it. 
Anyway, thanks for reporting, I'm taking a look at what's happening with latest toolchain on windows!

salvq, 2019-12-16:
Let me try and report again.

This is more what I found out just before you posted...

I just recompile original SDUBoot.ino without changing SS pin, transfer to h file and also Usage.ino compilation is failing due to above.

So the reason of compile issue of Usage.ino is either:

1. Compiling SDUBoot.ino (using latest Arduino IDE) or 
2. Transferring from bin to h file (using SDUBoot.ino.mkrgsm1400.bin | xxd -i > /tmp/mkrgsm1400.h).

- original size of mkrgsm1400.h 91 415 bytes
- recompiled size of mkrgsm1400.h with above approach 102 811 bytes

facchinm, 2019-12-16:
Could you attach the `mkrgsm1400.h` you generate? Thanks

salvq, 2019-12-16:
Attached, this is with SS PIN changed to 6, if neccessary I can generate original one (without any change)...

[mkrgsm1400.zip](https://github.com/arduino/ArduinoCore-samd/files/3966906/mkrgsm1400.zip)

salvq, 2019-12-16:
Just adding more tests I just did...

1. Compilation of Usage.ino in 1.6.21 board mng for both version of *.h compiled OK (original and compiled one from 1.8.3 board manager). Have not tried if it is actually working yet

2. Now I also tested to compile Usage.ino in 1.8.3 with compiled h file from 1.6.21 and it also works.

salvq, 2019-12-16:
I can confirm that point 2 is working (just tested with SD card), so the short temp fix is following:

1.) If anybody needs to recompile h include (i.e. mkrgsm1400.h), need to revert back to 1.6.2 board version and compile it there
2.) Then you can switch back to the latest board version 1.8.3 and compile using #include <SDU.h> in your script and should be read to go

Uup115, 2020-07-22:
Hello,

I also have the same problem in Linux.  Using latest board version 1.8.6.  I'm recompiling **mkrwifi1010.h**.  File size = 102 811 bytes.
Any idea when this might be fixed?