arduino/ArduinoCore-samd #487: Undefined "ADC_BATTERY" on MKRWAN1310

TechNyquist, 2020-01-15:
Hello everyone,

I believe this should be a simple one.
I developed my firmware on MKRWAN1300, now I moved to MKRWAN1310 because of claimed improvements in low-power consumption.

I have a snippet in my firmware that tries to detect battery level using the following 

`int sensorValue = analogRead(ADC_BATTERY);`

Now I moved the board to MKRWAN1310 and compiler says that "ADC_BATTERY" is undefined.
Looked in variant.h and actually there isn't any.

Aside: I searched around and found [this commit](https://github.com/arduino/ArduinoCore-samd/commit/206956dd4786a8cb5de8f8b179e0c5052689fb89) that reports that "ADC pin is not connected" on MKRWAN1310...dunno if this is related but it pinned up my attention.

So how should I test for battery level on MKRWAN1310 as ADC_BATTERY is undefined?

Thank you.

03sarath, 2020-03-16:
@TechNyquist you cannot read battery status using `ADC_BATTERY` because PIN 32 is now connected to SPI FLASH CHIP as `FLASH_CS` use any Analog Pins to read battery voltage.

TechNyquist, 2020-03-16:
I see. So a "manual" wiring to a generic  Analog Pin is required for this here.
Ok, thank you.

closed 2020-03-16 by TechNyquist