arduino/ArduinoCore-samd #491: MKR ZERO powering issue with battery

logicaprogrammabile, 2020-01-28:
Hello
by powering the mkrzero board with a 3.7v (3.0Ah) lipo battery, the ON LED is lightly powered. in the data sheet it is specified that the led ON is off when I use the battery.
This does not happen. This compromises the use of boards in low-power applications.
Could anyone check this out?

mweetman3, 2020-03-16:
I looked into this issue. The LED is slightly powered throught the USB protection circuit. The SAMD has ~40K internal pull up resistors to 3v3 on the USB P & N lines. Some of this current leaks through the USB protection device & end up on the VUSB supply. The current through the LED is around 40uA.
You can test this theory by shorting the USB P & N lines to GND. The LED then goes completely out.

sslupsky, 2020-03-16:
When you want to run in low power, you need to first detach the USB to turn it off.

USBDevice.detach();

mweetman3, 2020-03-17:
Thanks for that. I thought I had tried all the detach options!
This does indeed stop the LED from lighting, & saves 0.63mA from the battery current.
Going back to USB power, you simply need to:-
USBDevice.attach();
Serial.begin(115200);
Also to reconnect USB, you need to be aware of:-
https://github.com/arduino/ArduinoCore-samd/pull/352
THANKS again for the tip.

logicaprogrammabile, 2020-03-24:
i have tested the tips and i have seen the code turn the led off.
i have tested the code in two mkrzero boards and the led go off.
thank you for te tip

closed 2020-03-24 by logicaprogrammabile