arduino/ArduinoCore-samd #496: dpinst-amd64.exe is not signed for Windows 10

don, 2020-02-04:
dpinst-amd64.exe is not signed (or the signature has expired)

Steps to duplicate

- Windows 10
- Arduino IDE 1.8.11
- Open Boards Manager
- Install Arduino SAMD Boards version 1.8.4
- During the install you'll get an warning about unsigned drivers

[image]

Users can work around this by saying yes, but it's not OK to expect users to ignore this warning. Signed drivers are really important for avoiding malware. Arduino software installs shouldn't be encouraging people, especially beginners, to bypass security restrictions.

Suggested resolution: replace [dpinst-amd64.exe](https://github.com/arduino/ArduinoCore-samd/blob/master/drivers/dpinst-amd64.exe) with an updated version that has a valid signature.

cmaglie, 2020-02-04:
AFAIK those are the same dpinst coming directly from Microsoft...
By the way I'll try to create a signed version of those to try it out.

don, 2020-08-20:
@cmaglie I’m still seeing with Arduino 1.8.13 and SAMD 1.8.8 on a new Windows 10 machine. Can you revisit getting a newer signed version of dpinst-amd64.exe into the Arduino installer? The latest Windows Driver Kit (WDK) should have a newer signed version of the exe. Thanks!

nahshonw, 2022-09-12:
12 Sept 2022 and I had to google this issue on Windows 10.  Arduino is a big enough name to have signed drivers. I feel violated for saying yes on this clean install. Please just fix it.

dlimabean, 2022-11-26:
Still a problem on Nov 26 2022

dlimabean, 2022-11-26:
Are we confident with the distributors of arduino to ignore this problem?

cmaglie, 2022-12-02:
Finding where `dpinst.exe` is placed in the Microsoft Developer Kit is an epic quest on his own.

I've installed the latest WDK for Windows 11, but can't find `dpinst.exe` anywhere.

After an hour of google search, I finally learned that it is part of the **Driver Install Framework (DIFx)** https://learn.microsoft.com/en-us/windows-hardware/drivers/install/difx-guidelines and it seems to be deprecated:

> Starting in Windows 10 Version 1607 (Redstone 1), the Driver Install Frameworks (DIFx) tools (Difxapi.dll, Difxapp.dll, Difxappa.dll, and DPInst.exe) are deprecated and are no longer included in the WDK.

So to find the latest available version of `dpinst.exe` we should look inside a WDK for Windows < 10 v1607.
The bigger version available is the Microsoft WDK 8.0 (since the WDK 8.1 seems to have been removed from the download page https://learn.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads#step-2-install-the-wdk).

I will save you the pain to install and find `dpinst.exe` inside the WDK 8.0: the file is installed in `C:/Program Files (x86)/Windows Kits/8.0/redist/DIFx/dpinst/MultiLin/x64/dpinst.exe`.

Here is the file: [dpinst.zip](https://github.com/arduino/ArduinoCore-samd/files/10142184/dpinst.zip) and AFAIC the file is not signed.

At this point I'm not sure what's the best way forward...

cmaglie, 2023-01-20:
PnPUtil may be an alternative to dpinst:
https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/pnputil

cmaglie, 2023-02-03:
I've investigated the use of `pnputil.exe`, it's very similar to `dpinst`, it has basically the same features, but it won't prompt the user to gain admin privileges (the so-called UAC prompt, shown in @don's screenshot).

```
C:\Users\cmagl\Code\ArduinoCore-samd>pnputil /add-driver drivers\arduino-samd.inf /install
Utilità Plug and Play Microsoft

Aggiunta del pacchetto driver in corso:  arduino-samd.inf
Non è possibile aggiungere il pacchetto driver: Accesso negato.        <----- this means "Access denied."

Pacchetti driver totali:  1
Pacchetti driver aggiunti:  0

C:\Users\cmagl\Code\ArduinoCore-samd>
```

So we cannot use `pnputil` because the `post-install.bat` script is run as a non-privileged user by the Arduino IDE.

cmaglie, 2023-02-03:
Another interesting fact is that the `dpinst.exe` distributed with the packages are **actually signed** by Microsoft:

[image]

This means that the UAC dialog is displayed because the driver installation **requires administrator privileges** (and not because the dpinse.exe is not signed).

There is no way to avoid that. Even if you try to run Arduino IDE as admin, to begin with, this will trigger another UAC asking the user if it's ok to run the IDE as admin.

umbynos, 2023-02-03:
After this tentative, it seems clear that this is an approach that we cannot follow. Another way of doing this would be to develop our tool that visualize the UAC and launches itself either pnputil or dpinst once it has the privileges.

per1234, 2023-02-07:
Hi @Sajons. Please post your question to the Arduino Forum:

https://forum.arduino.cc/c/using-arduino/installation-troubleshooting/18

I'll be happy to help you over there.

william-vh, 2023-12-03:
Arduino Version 1.8.19. Still an issue.
Safe?

PeterGribbin, 2024-01-11:
This issue also affects Arduino IDE 2.2.1.
It prevents me using the Arduino nano esp32 on work computers for which I do not have admin credentials.