arduino/ArduinoCore-samd #497: Keyboard library modifying COM port on MacOS

tigoe, 2020-02-06:
When the Keyboard library is added to a sketch, it has adverse effects on COM port availability on MacOS.

* Testing with MacOS Mojave 10.14.6 and IDE 1.8.10 and 1.8.11
* Tested on MKRZero and Nano 33 IoT
* Tested with SAMD cores 1.8.1 through 1.8.4

Test code:
````
#include <Keyboard.h>
void setup() {
}
void loop{
}
````

When you add the Keyboard library, compile, and upload, the Arduino's COM port de-enumerates from the OS during upload as usual. But then it doesn't return. It increments. After upload, you get a COM port back, but the number is changed. For example, if the original COM port was /dev/cu.usbmodem141201, then the new one will be /dev/cu.usbmodem141202. If you then upload a sketch without the Keyboard library included, the COM port changes back to /dev/cu.usbmodem141201. 

The issue is reliably repeatable on different boards in the SAMD model. I tried a couple of Nano 33 IoTs, two MKRZeros, and a MKR1010.

facchinm, 2020-02-12:
Hi Tom,
this is expected given how OSX enumerates the serial ports. Since the algorithm is based on USB iSerial field and we change it every time a new USB functionality is added (to help Windows installing the correct driver) the name changes too. 
Closing as expected, please reopen if needed.

closed 2020-02-12 by facchinm

tigoe, 2020-02-12:
Expected perhaps, but inconsistent with the user experience of the library
in the past, and inconsistent with the library behavior on the32u4 boards.
I would suggest we find a way to provide a consistent behavior across all
the Arduino boards that can use this library if possible.