arduino/ArduinoCore-samd #521: Can't find from libraries manager

LuisCRSousa, 2020-05-11:
Can't find from libraries manager

AidenDawn, 2020-05-11:
The core install in the Arduino IDE is usually found from the boards manager under "Board:" in the tools menu. You're looking for the Arduino SAMD boards entry, assuming I've understood what you're trying to do.

If your looking for a local copy of the core either grab this library from github or after installing, find it in your OS specific location.

Hope that helps

LuisCRSousa, 2020-05-11:
Sorry I've not detailed the problem.
From a fresh arduino install I cant find these arduino samd boards libraries at boards manager.

matthijskooijman, 2020-05-11:
Are you sure you are looking in the *boards manager* and not the *library manager* as you wrote?

If so, can provide some more details (OS, IDE version number) and maybe make a screenshot of the board manager showing the problem? One other thing to try is to run arduino from the commandline (run `arduino_debug.exe` if you are on Windows) to see if there are any error messages there.

One thing I can imagine is that it fails to download the list of available boards and then does not show them.

LuisCRSousa, 2020-05-11:
ups sorry. I was so in auto pilot mode that I've been clicking at libraries manager thinking that it was boards. Sorry :)

closed 2020-05-11 by LuisCRSousa