arduino/ArduinoCore-samd #527: Cant install package

Onno-Dirkzwager, 2020-06-09:
Hi I installed the latest version of the ArduinoCore-samd package about 2 months ago and had no problems until a few hours ago.

Symptoms:
- Arduino IDE wont start
- After I remove the ArduinoCore-samd folder I can start the IDE
- Reinstalling the package seems to stop at 99% and I can press the Intall button to start over again. Never finishes. And I cant restart the IDE until I manualy remove the folder.

What I tried:
- removing everything Arduino related(that I know of) including ~/Library/Arduino15/
- reinstalling IDE 1.8.8 - 1.8.12
- rebooting between above steps

But no matter what I do I cannot reinstall the ArduinoCore-samd package whithout ending in the symptoms mentioned above.

jes030456, 2020-06-10:
I am having the exact same problem using the 1.8.12 IDE trying to install the SAMD package. I am also getting a java.lang.nullpointerexception error after attempting to install. 

[image]

Onno-Dirkzwager, 2020-06-10:
I see similarities in that I don't see the boards either.

But the difference is that my boards manager does not show installed.

@jes030456 does your IDE start normally?

jes030456, 2020-06-10:
My board manager only shows installed immediately after the installation attempt. If I exit out of the window it looks as if I never installed it (but if I click install again an error message shows up saying file already exist). 

[image]

It does not start normally after installation, likewise I have to delete the entire content of the ArduinoData folder in order to open up the IDE again. (otherwise it does not open after printing "initializing package"). I am thinking there may be some settings that can be changed in the preferences.txt file that may fix the start up problem. 

Any idea?

facchinm, 2020-06-10:
We had a problem on our package_index.json that caused the error and prevented the IDE to start.
Now the package_index has been fixed, but to recover a working installation you must delete once again the package_index.json with the cache folder and restart the IDE as usual.

Windows:
* remove file C:\Users\YourUsername\AppData\Local\Arduino15\package_index.json
* remove folder C:\Users\YourUsername\AppData\Local\Arduino15\cache

Mac:
* remove file /Users/YourUsername/Library/Arduino15/package_index.json
* remove folder /Users/YourUsername/Library/Arduino15/cache

Linux:
* remove file /home/YourUsername/.arduino15/package_index.json
* remove folder /home/YourUsername/.arduino15/cache

Please note that the problem that prevents the IDE to start has already been fixed and it's ready for the next release, that's the reason why the Nightly/Beta Builds are not affected, and also the reason why we did not detect this problem earlier, sorry about that!

closed 2020-06-10 by facchinm

Onno-Dirkzwager, 2020-06-10:
@facchinm 
I can confirm all is back to normal. 
Thankyou for the support!

printedman, 2020-06-11:
Me ok too. Thanks