arduino/ArduinoCore-samd #538: Lost characters in Serial.print with many Interrupts in background

AchimErk, 2020-06-28:
Hi,
I'm logging to SERIAL and have a lot (~3000 per sec) of iterrupts. 

When having the interrupts active, the SERIAL.print() often drops characters (return 0 bytes written). This was reproducable on MKRzero and on MKR1000.

Fix is a small change in file USBCore.cpp:
When checking for last send was finished, its first tested for IsReady, then it waitet for TransferComplete. My assuption is, that ISR is happening inbetween, so there is a bigger time difference between both lines. 

Adding in the wait loop the condition for IsReady solved the issue.

Here the code:
uint32_t USBDeviceClass::send(uint32_t ep, const void *data, uint32_t len)
{
	[...]
	while (len != 0)
	{
		if (usbd.epBank1IsReady(ep)) {
			// previous transfer is still not complete
                        [...]
			// Wait for (previous) transfer to complete
			// inspired by Paul Stoffregen's work on Teensy
			// [Pae] old version : while (!usbd.epBank1IsTransferComplete(ep) && usbd.epBank1IsReady(ep)) {
			while (!usbd.epBank1IsTransferComplete(ep) && usbd.epBank1IsReady(ep)) {
			// [Pae]                                                         ^^^ ^^^^^^^^^^^^^^
				if (LastTransmitTimedOut[ep] || timeout-- == 0) {
					LastTransmitTimedOut[ep] = 1;

					// set byte count to zero, so that ZLP is sent
					// instead of stale data
					usbd.epBank1SetByteCount(ep, 0);
					return -1;
				}
			}
		}

What are your thoughts?
Can this be fixed?

Thanks and best regards
Achim

habazut, 2022-11-01:
The DCC-EX devel team has seen the same problem when trying to port to SAMD. At least I think so. Unfortunately the code example of the original poster seems to be garbled.

In DCC-EX we have the combination of interrupts every 58us and serial print which are very short, say 1 to 10 characters. Now and then the print do not result in any USB packet. With a traditional UART and USB chip this **never** happens. 

The code in the USB core has as I see it the following two problems:

1. `if (usbd.epBank1IsReady(ep))` is not enough.  One must look at both flags (bank ready and transfer complete) before the buffer should be touched again.

2. The timeout code is strange. Why so short? Would it not be better to wait longer (say some millisecond) so that the USB very likely gets it's work done? We are simlating a serial UART after all.

So the whole timeout code at the beginning should be replaced with something much simpler like

```
while (usbd.epBank1IsReady(ep) && !usbd.epBank1IsTransferComplete(ep)) {
  // optional timeout code here 
};
```

If the timeout code is replaced with this busy wait no dropped USB packets happen.

On Timeouts: Yes, there should be a timeout but it should be in the ms range and yes it can be done by a counter, for example by counting and onolooking every counter wrap att micros(). If a long timeout occurs it is probably smart to restart the whole USB stack as somebody might have yanked the USB cable or the like.

Regards,
Harald.

To reproduce the problem you need to checkout [ee4963d] (https://github.com/DCC-EX/CommandStation-EX/commit/ee4963dfca8cdc0f887262e868e77efd1ce6d458), build with `#define DIAG_LOOPTIMES` in `defines.h` and then watch the serial. https://github.com/DCC-EX/CommandStation-EX/

Neil-McK, 2023-01-24:
Adding to the above, I had also observed that the SAMD SerialUSB was very very slow (~15kBaud effective throughput in some cases).  Applying Harald's fix above now gives me nearer 275kBaud as well as fixing the lost character problem.

Incidentally, I had also noted that the Arduino IDE didn't display the board type against the serial port name, and the Tools / GetBoardInfo menu option failed.  Now both of these work correctly.

Ash-4, 2023-04-25:
Perhaps this will help identify what needs to be changed.  Thanks.
[image]

habazut, 2023-04-28:
Suggested patch to USBCore.cpp
[USBCore.cpp.CRLF.txt](https://github.com/arduino/ArduinoCore-samd/files/11356639/USBCore.cpp.CRLF.txt)