arduino/ArduinoCore-samd #542: failure in high speed communication

ROBOTIS-Will, 2020-07-09:
Hi,

This issue is an explanation of test result regarding the PR #541.
While we've been testing with Arduino MKR boards, there was an issue with unsuccessful transmission caused by the flushUART(). 

In our test we were using half duplex communication(TTL) over the UART to control TX and RX direction with the direction control GPIO pin. After flushing the UART, direction control will be switched, but the flushUART does not wait until the completion of TX.

Please see below oscilloscope graph where the direction control(in blue) drops too early before the end of TX(in green), therefore, the actual TX data is missing the last bit(in yellow).

[image]

After commenting out the isDataRegisterEmptyUART() section, flushUART() waits until the end of transmission as below.
[image]

As we are not sure the reason of skipping the checking transmission completion, please review this modification(#541) and let us know.
Thank you.

matedl, 2020-08-27:
I have the same problem when writing a single byte over half duplex RS-485. When I write multiple bytes, flush() works as it should but when I write a single byte it does not work. When writing a single byte "isDataRegisterEmptyUART()" probably returns true since the byte is already in the shift register and flush() returns without waiting for TXC. Maybe this problem was introduced when fixing the issue "Fix Serial.flush() blocking if called immediately after Serial.begin() #144"?

ROBOTIS-Will, 2020-09-21:
@facchinm Hi, I'm just checking in for any updates or progress on this issue. Hope you are safe from the Covid-19.

facchinm, 2020-12-18:
Patch #541 merged, will be part of core 1.8.11 :slightly_smiling_face:

closed 2020-12-18 by facchinm