arduino/ArduinoCore-samd #547: Tone lower than ~40 Hz doesn't work

NikilRagav, 2020-07-16:
I see no technical reason why this shouldn't work.

Assuming the generic timer clock is 8MHz, then tone() should be able to support as low a frequency as .11Hz with a 16 bit timer by my calculation

(my goal is to use it to flash an LED 3 times in one second, but it doesn't seem to work on my Sparkfun Qwiic Micro which is a SAMD21 board).

As per the formula in the function, at 3Hz, (8E6/2)/3/(2^6) < 0xFFFF, so it should work right? Am I making a calculation error?

Side note (Potentially adding an overflow counter could be pretty useful for longer tone durations)

NikilRagav, 2020-07-16:
Actually, doing some more investigation on this,

tone(1 /*Hz*/) works with any duration

tone(2) and higher seem to just result in the LED being constantly lit, no frequency at all

NikilRagav, 2020-07-16:
also a nice feature would be to be able to invert the order (high then low)

ZombieFodder, 2025-01-21:
This might be related to an issue I raised with Adafruit fork of this repo:
https://github.com/adafruit/ArduinoCore-samd/issues/371