arduino/ArduinoCore-samd #559: un-useful and bloaty assert() in SPI library.

WestfW, 2020-09-03:
The SPIClass Constructor in the SPI library contains an assert:

```
SPIClass::SPIClass(SERCOM *p_sercom, uint8_t uc_pinMISO, uint8_t uc_pinSCK, uint8_t uc_pinMOSI, SercomSpiTXPad PadTx, SercomRXPad PadRx) : settings(SPISettings(0, MSBFIRST, SPI_MODE0))
{
  initialized = false;
  assert(p_sercom != NULL);
  _p_sercom = p_sercom;
```

It turns out that assert() generates live code on samd processors, and the default (defined in newlib) function invokes fiprintf()
Since stdio isn't set up in most Arduino compiles, this is pretty useless, and it can cause code bloat in sketchs that include SPI.
(I think this is normally masked because there is only one SPI port on most SAMD platforms, so the constructor always optimizes away the constructor.  But it's an issue for boards with more than one SPI.  See https://forums.adafruit.com/viewtopic.php?f=62&t=169034 )

aentinger, 2020-10-05:
Hi @WestfW :wave: care to create a PR to remove it?

WestfW, 2020-10-08:
Ok.