arduino/ArduinoCore-samd #561: fix your makefile

linuxgit1, 2020-09-22:
The makefile inside the /bootloaders/zero directory doesn't work properly.

The INCLUDES don't take into account the header files inside that directory (such as board_driver_i2c.h) since it gives an error
when building.

husztikd, 2020-09-24:
I can see the same. I tried to compile master, 1.8.5 and 1.8.6 and got the same error: 

board_driver_i2c.h:23:10: fatal error: sam.h: No such file or directory (with the sam.h some subdirs away)

closed 2020-09-28 by linuxgit1

matthijskooijman, 2020-09-28:
@linuxgit1 for future reference, it's better  to leave a comment saying that you solved it, rather than editing your original post, since:
 - People get notified about comments, not edits
 - Now your original post is gone, which makes this issue rather useless for anyone else running into the same problem (though @husztikd's comment has some of the same info, I guess).

linuxgit1, 2020-09-30:
@matthijskooijman I understand your point. However, since someone changed the title such that it reflects my original main problem, I edited my comment such that it contains the solution in order to help people in the future that have the same problem as stated in the title.