arduino/ArduinoCore-samd #563: regression: OTA updates no longer working on Wifi1010 & SAMD M0 release 1.8.8

Tdegussem, 2020-09-29:
When trying to do an Over-the-Air update on a maker board (Wifi1010), the board hangs after OTA flashing:

ArduinoOTA example,  do ethernet port upload using OTA 1.8.8:
- IDE shows successfull flashing 'OK - Sketch uploaded successfully' --> Auto-Reset --> Board is unresponsive (No COM port, no USB hotplugging)
ArduinoOTA example,  do ethernet port upload using OTA 1.8.7:
- IDE shows successful flashing 'OK - Sketch uploaded successfully' --> Auto-Reset --> Board is functional

I can reproduce this by up- or downgrading the board definition version.

jandrassy, 2020-10-04:
additional information. the cause is 
`mkrwifi1010.arduinoota.extraflags=-d`
in boards.txt

this is for the 'new' OTA. does it work? is it documented?

facchinm, 2020-10-05:
Hi @jandrassy ,
the `-d` flag was meant to allow offloading the download to Nina module (https://github.com/arduino/arduinoOTA/blob/master/main.go#L39) .
The problem is that, for security reasons, the second stage bootloader now requires an image with a particular header, so passing the raw binary will not work.
@aentinger should we revert the `-d` or adapt `arduinoota` to create and serve the correct binary?

jandrassy, 2020-10-05:
@facchinm Martino, right now here the problem is a conflict with the WIFi101OTA library. Until the new OTA download is ready and documented there shouldn't be the `mkrwifi1010.arduinoota.extraflags=-d` line in boards.txt.

It looks like the file is downloaded by firmware from arduinoOTA server at the same time as it is uploaded by arduinoOTA to WiFi101OTA server. But I don't know yet why it is a problem.

aentinger, 2020-10-05:
Hi :wave: I've got next to no idea (really no idea actually) how OTA via IDE works. However, [here](https://github.com/arduino-libraries/ArduinoIoTCloud/tree/master/extras/tools) are the tools to prefix the binary with the required data. I should note however that those requires using [ArduinoCore-samd:master:HEAD](https://github.com/arduino/ArduinoCore-samd/tree/9f92beeecd6d29381be3cb398432aafabf198ded) since there's been no release since the latest changes to **SNU**. What's the fastest strategy to get to a fix here?

closed 2020-10-05 by facchinm

jandrassy, 2020-10-05:
that was fast :-). thank you

aentinger, 2020-10-05:
Yeah @jandrassy ... a release was already in the air :cloud_with_lightning: :wink:

PR #565 (2020-10-05): Temporary disable OTA download via nina ...
... until the new header structure has been implemented in all the necessary tools.

This kind of fixes #563.

@facchinm How should we proceed so we don't forget to repair the regression?