arduino/ArduinoCore-samd #571: Building bootloader for MKR GSM 1400 and MKR NB 1500

ksmith3036, 2020-11-22:
I have built the bootloaders from this repo, with source code current as of now.
Have been using make of Cygwin and also tried using Atmel/Microchip Studio. I guess there is no differencem since both really uses the toolset that come with the Arduino support for the SAMD21 boards.

The Arduino Zero builds fine, but the MKR GSM 1400 and MKR NB 1500 fails when compiled with DEBUG enabled or the end bootloader files is about 1 KByte smaller than the files installed by Arduino studio.

This build fails:
`$ make DEBUG=1 SAM_BA_INTERFACES=SAM_BA_BOTH_INTERFACES  BOARD_ID=arduino_mkrnb1500 NAME=samd21_sam_ba_arduino_mkrnb1500 clean all`
**Error: arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 1156 bytes**
[Build error MKR NB 1500.txt](https://github.com/arduino/ArduinoCore-samd/files/5578765/Build.error.MKR.NB.1500.txt)

This build works:
`$ make DEBUG=1 SAM_BA_INTERFACES=SAM_BA_USBCDC_ONLY  BOARD_ID=arduino_mkrnb1500 N00 NAME=samd21_sam_ba_arduino_mkrnb1500 clean all`

The built file, samd21_sam_ba_arduino_mkrnb1500.bin is 8164 bytes, almost matching the bootloader supplied by Arduino: C:\Users\kaypro\AppData\Local\arduino15\packages\arduino\hardware\samd\1.8.9\bootloaders\mkrnb1500\samd21_sam_ba_arduino_mkrnb1500.bin of 7980 bytes size.
[Build only CDC with debug MKR NB1500.txt](https://github.com/arduino/ArduinoCore-samd/files/5578767/Build.only.CDC.with.debug.MKR.NB1500.txt)

If bulding without debug, I can build for both USB and serial:
`$ make SAM_BA_INTERFACES=SAM_BA_BOTH_INTERFACES  BOARD_ID=arduino_mkrnb1500 NAME
=samd21_sam_ba_arduino_mkrnb1500 clean all`
[Build no debug - both serial and USB  - MKR NB 1500.txt](https://github.com/arduino/ArduinoCore-samd/files/5578770/Build.no.debug.-.both.serial.and.USB.-.MKR.NB.1500.txt)

But it is quite suspicious that the resulting samd21_sam_ba_arduino_mkrnb1500.bin is only 6544 bytes long, as the prebuilt Arduino supplied file is 7980 bytes long.

Exactly the same goes for the MKR GSM 1400 build, since I guess only the VID and PIDS is changed from the NB 1500 board.

So my questions are these:
1. The bootloader files supplied by Arduino, are they compiled for both serial and USB, or only USB?
2. The bootloader files supplied by Arduino, are they compiled with DEBUG or not?
3. If compiled for both serial and USB and without debug, why is my resulting files som much smaller than the prebuilt files supplied from Arduino?
4. Is the small 6544 bytes long file OK, so that I safely could flash it using a modified SAMD_BootloaderUpdater library?

When opening the prebuilt bootloader files and the bootlader files I built without DEBUG in an hex editor, I see that the structure is quite different. Look at the dumps of the hex editor, showing a very different placement of the versions trings and text information.
So both the size and structure of the file is suspicious.
[image]
[image]

I attached the build logs, since you then can verifiy that the correct options and toolset has been used.

facchinm, 2020-11-23:
Hi @ksmith3036 ,
the bootloaders are compiled via [this](https://github.com/arduino/ArduinoCore-samd/blob/master/bootloaders/zero/build_all_bootloaders.sh) script, with SAM_BA_BOTH_INTERFACES and no debug. 
Enabling debug may produce a binary bigger than 8KB (especially on platforms with a PMIC which must be configured at startup).
If you still want to debug the bootloader, either disable the pmic or modify the core linker script to relocate itself at 16K boundary (this will give you plenty of space to compile with unoptimized flags).
Closing for now even if the conversation can continue if you need more guidance.

closed 2020-11-23 by facchinm

ksmith3036, 2020-11-23:
> Hi @ksmith3036 ,
> the bootloaders are compiled via [this](https://github.com/arduino/ArduinoCore-samd/blob/master/bootloaders/zero/build_all_bootloaders.sh) script, with SAM_BA_BOTH_INTERFACES and no [...]

Thank you for your answer. 

I want to compile the same way as the supplied bootloaders is built. I just testet different ways, to try to get a bootloader with about the same size, to see if it looked allrigt, since flashing a non-working bootloader bricks the Arduino.

In effect, what I do is to build the MKR GSM 1400 and NB 1500 bootloaders exactly the same way as the script does it.
My worries come from the fact that the built bootloaders is way smaller than the supplied ones:
```
timestamp                     size    filename
23.11.2020  14.30             6 544 samd21_sam_ba_arduino_mkrgsm1400.bin
23.11.2020  14.29             6 544 samd21_sam_ba_arduino_mkrnb1500.bin
```
Here are the dates and file sizes of the prebuilt files that come with the SAMD board support downloaded through Android IDE:
```
timestamp                     size    filename
03.09.2019  07.06             7 980 samd21_sam_ba_arduino_mkrnb1500.bin
03.09.2019  07.06             7 984 samd21_sam_ba_arduino_mkrgsm1400.bin
```
Since the file sizes of my build, using the toolset that come together with the SAMD board support download, is about 1,% KBytes smaller, I guess something has changed.

facchinm, 2020-11-23:
Very likely it has been built with the old toolchain since we don't update them if really necessary. About the size it could be due to `lto` optimizations that improved significantly in the latest gcc releases.
About testing a self compiled bootloader I strongly advise getting an atmel ICE (or any other SWD programmer), solder the SWD pins on the bottom of the board and flash with no pressure to brick anything :slightly_smiling_face:

ksmith3036, 2020-11-23:
OK, I downloaded the package for the toolchain 4.8.3-2014q1 and downloaded CoreArduino-samd tag 1.6.21 sources. Now it builds without errors and the file size is nearly the same. Just 8 bytes diff! :-)
When building the latest sources with the 4.8.3-2014q1, I got a size diff of 100 bytes.

Not that I would like to build based on old sources, but since I haven't got the SWD programmer, I just wanted to be sure.

Good to hear that the new toolchain should work. 1,5 KByte less size for a 8KByte originally is very significant!

Keep up the good work!

One suggestion for improvement, when packaging the Arduino SAMD Boards (32-bits ARM Cortex-M0+) packages for download, is to document what source code level the prebuilt bootloaders have, because it seems that the prebuilt bootloaders is older than rest of the package.

Thank you!

ksmith3036, 2020-11-23:
I ended up building modified bootloaders, based on toolchain 4.8.3-2014q1 and CoreArduino-samd tag 1.6.21 sources. 
The purpose was to alter the behaviour with regards to issue  #570. 
The newly built bootloader was flashed using a modified SAMD_BootUpdater sketch.

And worked great!