arduino/ArduinoCore-samd #575: SAMD21 USBDeviceClass::ISRHandler() called continuously.

studiohsoftware, 2020-12-14:
Running the USBMIDI library on MKR1000, I discovered with a debug statement in USBDeviceClass::ISRHandler() within USBCore.cpp, the handler is called continuously. 

Specifically, adding Serial1.println("Handler Called"); to the first line of ISRHandler() exposed the problem. 

I did not look deeper to find the specific flag in DEVICE.INTFLAG causing the problem, and whether or not that flag is required for enumeration. However, executing this statement after enumeration seems to solve the problem.
USB->DEVICE.INTENCLR.reg = 0xFF;

matthijskooijman, 2020-12-15:
What is "continuously"? I'd expect the handler might be called ever 1ms, when start-of-frame (SOF) is received, if that ISR is enabled? If so, that would be by design, I think? Or are you saying it is called really all the time, blocking the main loop completely?

studiohsoftware, 2020-12-15:
It is indeed the SOF. USBDeviceClass::ISRHandler() appears to explicitly use it only to toggle LEDs. 

However it must be the case that your enumeration code within the ISR depends on it implicitly because I do not get a valid device address without it. Perhaps you can disable it once your enumeration is done with it. 

I have found this crude method to disable within the sketch loop()
```
  if (USB->DEVICE.DADD.reg > 0){
    USB->DEVICE.INTENCLR.reg = 0x04;
  }
```