arduino/ArduinoCore-samd #578: Header file WInterrupts.h not found after update to 1.8.10 (32-bits ARM Cortex-M3)

tangram67, 2020-12-17:
Hi to all.

After upgrading to 1.8.10 (32-bits ARM Cortex-M3) for an Arduino MKR NB 1500 board under Ubuntu 18.04 the header WInterrupts.h is no longer found:

/home/dirk/Arduino/libraries/Arduino_Low_Power/src/samd/ArduinoLowPower.cpp:4:10: fatal error: WInterrupts.h: No such file or directory
 #include "WInterrupts.h"

I'm using the Low Power Library by including:
#include "ArduinoLowPower.h"

Regards,
Dirk.

facchinm, 2020-12-17:
Hi @tangram67 ,
thanks for reporting. I pushed a fix here https://github.com/arduino-libraries/ArduinoLowPower/pull/35 . Would you mind testing and reporting if it works fine?

tangram67, 2020-12-17:
Hi @facchinm 

Yes the error has gone now. Compilation works fine now.

Thanks!

closed 2020-12-17 by aentinger