arduino/ArduinoCore-samd #580: Error while writing flash memory - Arduino Nano 33 IoT

tessi, 2020-12-17:
Hi :)

I just received an original Arduino nano 33 IoT and wanted to test the wifi. The firmware check told me I need to update, and I followed the instructions to update the firmware. Instead of the nice success-message, I received the following

[image]

Used versions:

* Arduino GUI: 1.8.13
* WiFiNINA: 1.8.0
* WiFi101-FirmwareUpdater-Plugin: the bundled version and [version 0.10.12](https://github.com/arduino/WiFi101-FirmwareUpdater-Plugin/releases/tag/v0.10.12)
* macOS catalina (10.15.7)

The flashing failed multiple times even after restarting, reconnecting, re-flashing things multiple times.
The firmware-check now reports that it can't connect to the wifi module (it used to work before):

```
14:36:58.459 -> WiFiNINA firmware check.
14:36:58.459 -> 
14:36:59.172 -> Communication with WiFi module failed!
```

I am at a loss on how to continue debugging this. Do you have any hints?
How could I resolve this issue and is it even something I should report here?

Note: I would be happy to ignore this issue, but now I can't use wifi from normal sketches anymore, which is a little frustrating.

Damiasroca, 2020-12-17:
> Hi @tessi and @Damiasroca ,
> it looks like the issue has been caused by samd core 1.8.10 release. I'm moving the issue on that repo to ease keeping track of it. In the meantime, please revert to samd core 1.8.9 if you need to update the firmware. Will keep you posted as soon as the issue is solved 🙂

Downgrading indeed did the trick, Up and running again. Thank you

tessi, 2020-12-17:
[image]

Downgrading indeed worked, thank you for the super helpful and quick response @facchinm  and the fix in progress @aentinger  ❤️

closed 2020-12-22 by facchinm