arduino/ArduinoCore-samd #597: Serial.flush() blocks forever

jpmeijers, 2021-02-03:
Using the latest version of the SAMD core, doing a Serial.flush() will hang forever.

Problem can be reproduced using this sketch:
```
void setup() {
  SerialUSB.begin(115200);
  delay(2000);
  SerialUSB.println("Start");

  Serial.begin(9600);
  Serial.flush();

  SerialUSB.println("Setup end");
}

void loop() {
  SerialUSB.println("Loop");
  delay(1000);
}
```

Which will only print:
```
15:19:30.572 -> Start
```

Tracing the issue down takes me to `Serial.flush()`, which is defined in Uart.cpp. This calls `sercom->flushUART();` in SERCOM.cpp.

The previous versions of this core had the following code:
```
void SERCOM::flushUART()
{
  // Skip checking transmission completion if data register is empty
  if(isDataRegisterEmptyUART())
    return;

  // Wait for transmission to complete
  while(!sercom->USART.INTFLAG.bit.TXC);
}
```

The current version has the `if(isDataRegisterEmptyUART())` commented out:

https://github.com/arduino/ArduinoCore-samd/blob/0b60a79c4b194ed2e76fead95caf1bbce8960049/cores/arduino/SERCOM.cpp#L115-L116

I don't understand why this is commented out, as clearly the `while(!sercom->USART.INTFLAG.bit.TXC);` will never return if `flush()` is called when the TX buffer is already empty.

facchinm, 2021-02-03:
See https://github.com/arduino-libraries/MKRWAN/commit/049391822099dae9db67a9ad11f4a2005e745547 . This PR https://github.com/arduino/ArduinoCore-samd/pull/541 is the root cause, and it's not been analyzed well enough before merging. Sorry for that. Can we find a workaround to fix both the issues? @aentinger @cmaglie

aentinger, 2021-02-04:
Revert #541? (Sounds cheap but as I see it clearly a bug was introduced).

matthijskooijman, 2021-02-04:
I haven't looked closely, but I remember that on AVR, there was the problem that TXC is only set *after* all pending data was sent and cleared when new data was queued. This left the problem that on startup, when no data was written yet, no data was pending, but TXC was not set either, so just waiting for TXC in flush would wait indefinitely. From the SAMD datasheet, the TXC bit seems to behave the same here. I haven't looked close enough if this explains the problems in this issue, but I suspect it might.

On AVR, IIRC we solved this by simply keeping a variable that says whether or not data has been written at all and use that to bypass TXC if needed, I think the same approach would be valid here too.

acicuc, 2021-04-15:
I've made some changes that seems to have fixed this issue, but I'd appreciate if others could test. You can checkout the changes in the latest commit from my old PR #535  a744583bd5c55141939409e8327eba771c531f1f . It was a mistake to to push to that branch - sorry for the mess, but anyway the specific changes relating to this issue are clearly visible.

coffeemachine, 2021-09-10:
@facchinm this bug is critical in my view. In my case it causes RS485 library to hang (sendBreak calls flush). Like @aentinger I'd suggest to revert PR #541 and make a release based on 1.8.11 with just this fixed. I'd address the solution (@acicuc) in a separate release. What do you think?

acicuc, 2021-10-17:
@coffeemachine was @matthijskooijman suggestion that I implemented in a744583 helpful in your case?

closed 2021-10-20 by facchinm

WanaGo, 2022-08-08:
Was the fix (#652) for this rolled out for the Arduino Due (SAM) board library, or was it specific only to the SAMD ?
As the same issue happens on the Arduino Due.

acicuc, 2022-08-13:
@WanaGo, currently it doesn't look like it, but you may be able to implement a similar approach in the SAM core?

PR #652 (2021-10-20): fix Serial.flush() blocks forever #597
* The aynchronous nature of the DRE and TXC interrupt flags
  causes issues (lockups) when the TX DATA register is empty on start
  and a flush is issued. Simply looking at the DRE prior to
  waiting for TXC is insufficient because the data register
  may well be empty but the shift register could still contain
  data, in this case SERCOM::flushUART() would return before TXC
  has been raised thus before flushing is complete.
* bool added to SERCOM.h to indicate when it is ok for
  SERCOM::flushUART() to wait for the TXC flag. This flag is
  set when any  [... cut]