arduino/ArduinoCore-samd #606: AceTime example compilation is breaking checks

stonehippo, 2021-02-22:
Several of the most recent PRs are reporting broken checks. I took a look and it's not the code in the PR that's failing but the compilation of the ActTime examples. It seems that there's a error with the SAMD support in that library. From one of the check logs:

```
2021-02-22T21:35:32.0796794Z In file included from /home/runner/Arduino/libraries/AceTime/examples/HelloDateTime/HelloDateTime.ino:23:0:
2021-02-22T21:35:32.0798014Z /home/runner/Arduino/libraries/AceTime/src/AceTime.h:26:2: error: #error SAMD21 with arduino:samd >= 1.8.10 not supported, https:
2021-02-22T21:35:32.0799215Z  #error SAMD21 with arduino:samd >= 1.8.10 not supported, https://github.com/bxparks/AceTime/issues/45
2021-02-22T21:35:32.0799822Z   ^~~~~
2021-02-22T21:35:32.0800430Z In file included from /home/runner/Arduino/libraries/AceTime/src/ace_time/internal/ZoneRegistrar.h:10:0,
2021-02-22T21:35:32.0801351Z                  from /home/runner/Arduino/libraries/AceTime/src/AceTime.h:35,
2021-02-22T21:35:32.0802176Z                  from /home/runner/Arduino/libraries/AceTime/examples/HelloDateTime/HelloDateTime.ino:23:
2021-02-22T21:35:32.0803195Z /home/runner/Arduino/libraries/AceCommon/src/AceCommon.h:45:2: error: #error SAMD21 with arduino:samd >= 1.8.10 not supported, https:
2021-02-22T21:35:32.0804216Z  #error SAMD21 with arduino:samd >= 1.8.10 not supported, https://github.com/bxparks/AceCommon/issues/9
2021-02-22T21:35:32.0804816Z   ^~~~~
2021-02-22T21:35:32.0805352Z Using library AceTime at version 1.6 in folder: /home/runner/Arduino/libraries/AceTime 
2021-02-22T21:35:32.0806163Z Using library AceCommon at version 1.4.3 in folder: /home/runner/Arduino/libraries/AceCommon 
2021-02-22T21:35:32.0807078Z Using library Wire at version 1.0 in folder: /home/runner/.arduino15/packages/arduino/hardware/samd/1.8.11/libraries/Wire 
2021-02-22T21:35:32.0807628Z 
2021-02-22T21:35:32.0807962Z Error during build: exit status 1
```

This appears in all of the Compile Examples logs for PR #605, for example. It seems like the best course of action is to remove AceTime from the checks.

zfields, 2021-03-09:
Just echoing this observation to elevate. This issue has a high severity, because the entire repository has ground to a halt.

stonehippo, 2021-03-28:
Bueller? Bueller?

facchinm, 2021-03-29:
@per1234 I'd proceed by merging #607 (to fix the CI at least), then if we get some feedback for the patch series ( arduino/ArduinoCore-API#144 and friends) we can revert it.

closed 2021-03-29 by facchinm