arduino/ArduinoCore-samd #608: [NANO 33 IOT]: AnalogReadCorrection not working!

hasenradball, 2021-03-03:
I used the example sketch "CorrectADCResponse" with 10 bit and get the Correction of (0,0).

Then I used the following lines:

```
  // Set default correction values and enable correction
  analogReadResolution(10);
  analogReadCorrection(0, 0);
```

And then the NANO 33 IOT is not able to measure anything! 

[Read_ADC_Example.zip](https://github.com/arduino/ArduinoCore-samd/files/6079082/Read_ADC_Example.zip)

hasenradball, 2021-03-03:
What's going wrong?

hasenradball, 2021-03-04:
It seems that the code which handles the correction is wrong!
But there is nothing documented.

facchinm, 2021-03-09:
Hi @hasenradball ,
did you follow the instructions here https://github.com/arduino/ArduinoCore-samd/blob/master/libraries/SAMD_AnalogCorrection/examples/CorrectADCResponse/CorrectADCResponse.ino#L10-L22 ? Setting the gain as 0 will multiply all the values by 0, so it's ok to see nothing. The issue, if there's one, is indeed in `CorrectADCResponse` sketch

hasenradball, 2021-03-09:
Hi facchinm,

sure I read the instructions.

I testet it on the Arduino 33 IOT with 12 bit and with 10 bit.
- with the 12 bit I got the correction values like (33, 2111)
- with 10 bit I got the correction values like (0, 0)

But with your comments there is a misunderstanding in Gain- and Offset-correction and absolute Gain and Offset?!?
Maybe the spelling in the sketch is wrong.

What would be your proposal to get on?

hasenradball, 2021-03-09:
Here you find the Output for the 10 bit test:
```
ADC(3.3V) = 1019
   Gain = 2053, ADC(3.3V) = 751
   Gain = 2054, ADC(3.3V) = 644
   Gain = 2055, ADC(3.3V) = 601
   Gain = 2056, ADC(3.3V) = 589
   Gain = 2057, ADC(3.3V) = 545
   Gain = 2058, ADC(3.3V) = 601
   Gain = 2059, ADC(3.3V) = 1395

Readings after corrections
   ADC(GND) = 0
   ADC(3.3V) = 1024

==================

Correction values:
   Offset = 1
   Gain = 0

Add the next line to your sketch:
   analogReadCorrection(1, 0);

==================
```

hasenradball, 2021-03-09:
Here you find the Output of the 12 -bit Test:
```
ADC(GND) = 3
   Offset = 7, ADC(GND) = 3
   Offset = 8, ADC(GND) = 3
   Offset = 9, ADC(GND) = 2
   Offset = 10, ADC(GND) = 1
   Offset = 11, ADC(GND) = 2
   Offset = 12, ADC(GND) = 2
   Offset = 13, ADC(GND) = 2
   Offset = 14, ADC(GND) = 2
   Offset = 15, ADC(GND) = 1
   Offset = 16, ADC(GND) = 1
   Offset = 17, ADC(GND) = 2
   Offset = 18, ADC(GND) = 2
   Offset = 19, ADC(GND) = 1
   Offset = 20, ADC(GND) = 1
   Offset = 21, ADC(GND) = 0

Gain correction
   Gain = 2048, ADC(3.3V) = 4058
   Gain = 2049, ADC(3.3V) = 4060
   Gain = 2050, ADC(3.3V) = 4061
   Gain = 2051, ADC(3.3V) = 4063
   Gain = 2052, ADC(3.3V) = 4063
   Gain = 2053, ADC(3.3V) = 4067
   Gain = 2054, ADC(3.3V) = 4070
   Gain = 2055, ADC(3.3V) = 4072
   Gain = 2056, ADC(3.3V) = 4073
   Gain = 2057, ADC(3.3V) = 4075
   Gain = 2058, ADC(3.3V) = 4076
   Gain = 2059, ADC(3.3V) = 4078
   Gain = 2060, ADC(3.3V) = 4079
   Gain = 2061, ADC(3.3V) = 4081
   Gain = 2062, ADC(3.3V) = 4081
   Gain = 2063, ADC(3.3V) = 4083
   Gain = 2064, ADC(3.3V) = 4085
   Gain = 2065, ADC(3.3V) = 4087
   Gain = 2066, ADC(3.3V) = 4088
   Gain = 2067, ADC(3.3V) = 4088
   Gain = 2068, ADC(3.3V) = 4093
   Gain = 2069, ADC(3.3V) = 4093
   Gain = 2070, ADC(3.3V) = 4094
   Gain = 2071, ADC(3.3V) = 4094
   Gain = 2072, ADC(3.3V) = 4094
   Gain = 2073, ADC(3.3V) = 4094
   Gain = 2074, ADC(3.3V) = 4094
   Gain = 2075, ADC(3.3V) = 4094
   Gain = 2076, ADC(3.3V) = 4094
   Gain = 2077, ADC(3.3V) = 4093
   Gain = 2078, ADC(3.3V) = 4094
   Gain = 2079, ADC(3.3V) = 4094
   Gain = 2080, ADC(3.3V) = 4094
   Gain = 2081, ADC(3.3V) = 4094
   Gain = 2082, ADC(3.3V) = 4094
   Gain = 2083, ADC(3.3V) = 4094
   Gain = 2084, ADC(3.3V) = 4094
   Gain = 2085, ADC(3.3V) = 4094
   Gain = 2086, ADC(3.3V) = 4094
   Gain = 2087, ADC(3.3V) = 4094
   Gain = 2088, ADC(3.3V) = 4094
   Gain = 2089, ADC(3.3V) = 4095
   Gain = 2090, ADC(3.3V) = 4094
   Gain = 2091, ADC(3.3V) = 4095
   Gain = 2092, ADC(3.3V) = 4094
   Gain = 2093, ADC(3.3V) = 4095
   Gain = 2094, ADC(3.3V) = 4095
   Gain = 2095, ADC(3.3V) = 4094
   Gain = 2096, ADC(3.3V) = 4095
   Gain = 2097, ADC(3.3V) = 4095
   Gain = 2098, ADC(3.3V) = 4095
   Gain = 2099, ADC(3.3V) = 4095
   Gain = 2100, ADC(3.3V) = 4094
   Gain = 2101, ADC(3.3V) = 4095
   Gain = 2102, ADC(3.3V) = 4095

Readings after corrections
   ADC(GND) = 1
   ADC(3.3V) = 4095

==================

Correction values:
   Offset = 21
   Gain = 2089

Add the next line to your sketch:
   analogReadCorrection(21, 2089);

==================
```

hasenradball, 2021-03-10:
Hi facchinm,

one question beside.
How is the behaviour using a arduino IDE from zip (from hourly builds) in parallel to a fixed installation of the IDE?
I pointed out if I use the zipped IDE and install a new board it is installed in the path of the fixed installed IDE and not in the folder where I unpack the zip.
Does the Zip IDE use all settings from the fixed installed IDE?

hasenradball, 2021-03-10:
Or is it intented to copy the content of the zip into the installed folder to be up to date?

hasenradball, 2021-03-20:
Hi facchinm,

did my feedback help.
Could you have a look at it?
I s there an failure in the sketch and maybee in the docu of signals?

facchinm, 2021-03-29:
@hasenradball thanks for reporting back and sorry for the delay.
I'd keep the code as is and just document that is only works for the default analog resolution.
@per1234 can you take care of wording it better than I could ever do :slightly_smiling_face: ? Thx

hasenradball, 2021-03-29:
Thanks for your reply,

why does it only work for the default resolution?

Thanks