arduino/ArduinoCore-samd #616: Builds broken in master branch (1.8.11)?!

henrygab, 2021-04-01:
Is something misconfigured for the CI builds?

It appears that all PRs since December 24 have failed to build (via CI Actions).

On December 24th, three PRs (#3, #586, #588) were merged, and then the version was updated to 1.8.11.  Each of those merges has failed the CI build (but logs are gone now).

Please fix so PRs can have some sanity checks applied, and be considered for merging again!

Potentially affected PRs include:

* #613 - Fix compiler warning from useless volatile 
* #612 - Don't spill arduino namespace in headers 
* #610 - fix: address warning 
* #605 - Update Wire peripheral handler 
* #602 - Update HID.h with report types 
* #600 - Add a weak attribute to serial IRQ allowing real time communication lib. 
* #592 - Improve on bootloader security when Security Bit is set
* #591 - Fix I2C Master hang when communicating with bad I2C slaves. 
* #590 - Slave Wire: fix TwoWire::onService() to handle repeated start

facchinm, 2021-04-01:
Hi @henrygab ,
there's a known incompatibility on branch 1.8.11+ with this library ( https://github.com/bxparks/AceTime ). The check has been added to avoid future breakages but until https://github.com/bxparks/AceTime/pull/57 and https://github.com/arduino/ArduinoCore-samd/pull/612 get merged the CI will be "broken" (as expected).
@per1234 do we have any solution beside manually checking the CI output to make sure only AceTime fails to compile?

per1234, 2021-04-01:
@facchinm unfortunately there is no established system in place for doing that. The compilation results are all recorded in a JSON file, so it could certainly be done.

I can make the manual checking less painful though. I am currently preparing a PR to disable the verbose output from the `arduino/compile-sketches` action. The verbose output is primarily intended to be used for troubleshooting the build process or action, and really doesn't offer any value for normal usage. Disabling the verbose output will significantly reduce the size of the log and make it easier for people to check for the cause of the failure.

stonehippo, 2021-04-12:
@per1234 @facchinm any joy? It looks like now @bxparks is holding the PR on the AceTime side until 1.8.12 is released here? https://github.com/bxparks/AceTime/pull/58

facchinm, 2021-04-12:
https://github.com/arduino/ArduinoCore-samd/issues/597 needs to be fixed (and https://github.com/arduino/ArduinoCore-samd/pull/622 merged) before releasing 1.8.12, but all the other pieces should be in place now.

stonehippo, 2021-05-25:
Any progress?

bxparks, 2021-11-04:
I was going to create a PR to remove AceTime from the GH Workflow, but it was already done a few weeks ago:
https://github.com/arduino/ArduinoCore-samd/commit/96ce5f960191a20aeaca1858115111a9ecb9b4e1