arduino/ArduinoCore-samd #644: Uart buffer data loss leading to hang on MKRNB1500

joshbober, 2021-09-10:
Hello. We have been looking into this issue [on the MKRNB repo](https://github.com/arduino-libraries/MKRNB/issues/86), but I believe we have traced it down to the SAMD core being the root cause. 

In the Modem.cpp file of the MKRNB library, part of the code waits for an AT response from the cellular modem that communicates through a uart interface. Once the uart buffer has data (uart->available() returns nonzero), each character gets stored in a local variable _buffer. This buffer eventually gets checked for an expected result code that ends every AT response. The issue is, for responses longer than 63 characters, the buffer only stores the first 63 and drops the rest. This means it also drops the result code, and the function assumes that it is still waiting for the end of the response. 

This behavior occurs in versions 1.8.10 and 1.8.11 of the SAMD core. Reverting to version 1.8.9 resolves the issue. Please let me know what points need clarification, as I'm sure I haven't explained everything thoroughly. More details can be found in the link to the issue at the top of this comment. 

I am happy to help with testing to resolve this if needed. Thank you!

CptHolzschnauz, 2021-09-12:
I think this is a serious tricky bug to track down! 
Due to it's nature i guess the bug affects a lot of modem / AT console operations and can be a explonation for a LOT of puzzling errors (as described by many in the forum) on the mkr1500 & 1400.
The "Receive SMS" example sketch is a secure way to reproduce the issue.

joshbober, 2021-11-18:
I believe this is now fixed by #651. I have tested the ReceiveSMS example sketch on the MKR NB1500 with version 1.8.12 of the SAMD core. The device no longer hangs, multiple messages can be received consecutively.

closed 2021-11-18 by joshbober