arduino/ArduinoCore-samd #654: 1.8.12 changed name/space for TwoWire?

ladyada, 2021-11-17:
hiya still digging into what happened here, but with the new release, our I2C libraries no longer recognize TwoWire as a type

https://github.com/adafruit/Adafruit_Seesaw/runs/4241973792?check_suite_focus=true#step:6:3947

jandrassy, 2021-11-17:
this commit could be the problem:
https://github.com/arduino/ArduinoCore-samd/commit/a3cc04d0ce851819e6619ebf6947826b5949a7c4#diff-3327f28ed0d534a62c42e2630089372fcff9a90007e9523abb8128bcc666f3e2

facchinm, 2021-11-18:
Hi @ladyada ,
nothing intentional of course, but the patch linked by @jandrassy shouldn't have any side effect for TwoWire visibility.
In fact, I'm compiling locally with 1.8.12 the basic example from Adafruit_Crickit and it doesn't break (with Java IDE) :thinking: 
```
/ssd/Arduino-master/build/linux/work/arduino-builder -compile -logger=machine -hardware /ssd/Arduino-master/build/linux/work/hardware -hardware /home/martino/.arduino15/packages -hardware /home/martino/eslov-sk/hardware -tools /ssd/Arduino-master/build/linux/work/tools-builder -tools /ssd/Arduino-master/build/linux/work/hardware/tools/avr -tools /home/martino/.arduino15/packages -built-in-libraries /ssd/Arduino-master/build/linux/work/libraries -libraries /home/martino/eslov-sk/libraries -fqbn=arduino:samd:arduino_zero_native -ide-version=10817 -build-path /tmp/arduino_build_183778 -warnings=all -build-cache /tmp/arduino_cache_105633 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/martino/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/home/martino/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.openocd.path=/home/martino/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=/home/martino/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.CMSIS-Atmel.path=/home/martino/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=/home/martino/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -prefs=runtime.tools.CMSIS.path=/home/martino/.arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=/home/martino/.arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.bossac.path=/home/martino/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=/home/martino/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.arduinoOTA.path=/home/martino/.arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=/home/martino/.arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -verbose /home/martino/eslov-sk/libraries/Adafruit_seesaw_Library/examples/Crickit/basic/basic.ino
Using board 'arduino_zero_native' from platform in folder: /ssd/Arduino15/packages/arduino/hardware/samd/1.8.12
Using core 'arduino' from platform in folder: /ssd/Arduino15/packages/arduino/hardware/samd/1.8.12
Detecting libraries used...
/home/martino/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10817 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Zero\"" -I/home/martino/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/home/martino/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/ -I/ssd/Arduino15/packages/arduino/hardware/samd/1.8.12/cores/arduino/api/deprecated -I/ssd/Arduino15/packages/arduino/hardware/samd/1.8.12/cores/arduino/api/deprecated-avr-comp -I/ssd/Arduino15/packages/arduino/hardware/samd/1.8.12/cores/arduino -I/ssd/Arduino15/packages/arduino/hardware/samd/1.8.12/variants/arduino_zero /tmp/arduino_build_183778/sketch/basic.ino.cpp -o /dev/null
Alternatives for Adafruit_Crickit.h: [Adafruit_seesaw_Library@1.5.4]
ResolveLibrary(Adafruit_Crickit.h)
  -> candidates: [Adafruit_seesaw_Library@1.5.4]
/home/martino/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10817 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Zero\"" -I/home/martino/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/home/martino/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/ -I/ssd/Arduino15/packages/arduino/hardware/samd/1.8.12/cores/arduino/api/deprecated -I/ssd/Arduino15/packages/arduino/hardware/samd/1.8.12/cores/arduino/api/deprecated-avr-comp -I/ssd/Arduino15/packages/arduino/hardware/samd/1.8.12/cores/arduino -I/ssd/Arduino15/packages/arduino/hardware/samd/1.8.12/variants/arduino_zero -I/home/martino/eslov-sk/libraries/Adafruit_seesaw_Library /tmp/arduino_build_183778/sketch/basic.ino.cpp -o /dev/null
Alternatives for Adafruit_I2CDevice.h: [Adafruit_BusIO@1.9.4]
ResolveLibrary(Adafruit_I2CDevice.h)
  -> candidates: [Adafruit_BusIO@1.9.4]
/home/martino/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10817 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON "-DUSB_MANUFA [... cut ...]
Using library SPI at version 1.0 in folder: /ssd/Arduino15/packages/arduino/hardware/samd/1.8.12/libraries/SPI 
/home/martino/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-size -A /tmp/arduino_build_183778/basic.ino.elf
Sketch uses 17068 bytes (6%) of program storage space. Maximum is 262144 bytes.
Global variables use 3628 bytes (11%) of dynamic memory, leaving 29140 bytes for local variables. Maximum is 32768 bytes.
```
Are you using by any chance any special flag in the cli script?

facchinm, 2021-11-18:
Ok, so I reverted BusIO to 1.9.3 and reproduced the bug. In fact, it's caused by `Adafruit_I2CDevice.h` only including `Wire.h`, and this thing cascades into the missing namespace specification

My hint is to simply add `#include "Arduino.h"` at the top of the file since it shouldn't have any relevant side effect in any 
platform

```patch
diff --git a/Adafruit_I2CDevice.h b/Adafruit_I2CDevice.h
index 28468ea..7748e99 100644
--- a/Adafruit_I2CDevice.h
+++ b/Adafruit_I2CDevice.h
@@ -1,3 +1,4 @@
+#include <Arduino.h>
 #include <Wire.h>
 
 #ifndef Adafruit_I2CDevice_h
[martino@arch-elitebook Adafruit_BusIO]$ git diff > /tmp/busIO_1.8.3.patch
[martino@arch-elitebook Adafruit_BusIO]$ cat /tmp/busIO_1.8.3.patch
diff --git a/Adafruit_I2CDevice.h b/Adafruit_I2CDevice.h
index 28468ea..7748e99 100644
--- a/Adafruit_I2CDevice.h
+++ b/Adafruit_I2CDevice.h
@@ -1,3 +1,4 @@
+#include <Arduino.h>
 #include <Wire.h>
 
 #ifndef Adafruit_I2CDevice_h
```

ladyada, 2021-11-18:
we did find an easy fix, turns out you just need to #include <Arduino.h> before including Wire.h

https://github.com/adafruit/Adafruit_BusIO/commit/9a525b2ba3315d44640030830620cc4bf085c569

maybe Wire.h can add that include? this only affects libraries but i think there could be a large number of libraries that include Wire.h but dont include Arduino.h

ladyada, 2021-11-18:
wait, thats what you just said. lol I AGREE!@

facchinm, 2021-11-19:
LOL :slightly_smiling_face: In fact we could indeed add the include to Wire.h directly but you never know the side effects (like breaking libraries that forward declare classes https://github.com/bxparks/AceTime/pull/57 ).
Anyway, if we get some more breakages (I doubt that but you never know) we can definitely go for that solution.
Thanks again for reporting and sorry for the inconvenience

closed 2021-11-19 by facchinm