arduino/ArduinoCore-samd #660: Error Writing into flash memory - Arduino Nano 33 IoT

sntmasibon, 2022-01-16:
Hi! I'm a beginner in Arduino Nano 33 IOT. Before I use a wemos d1 mini and I used to save some data, in the EEPROM Memory, (EEPROM.write), a word that I save byte to byte. I saw a few of examples from the library <FlashStorage.h>, one of them consists on this:
`#include <FlashStorage.h>

// Create a structure that is big enough to contain a name
// and a surname. The "valid" variable is set to "true" once
// the structure is filled with actual data for the first time.
typedef struct {
  boolean valid;
  char name[100];
  char surname[100];
} Person;

// Reserve a portion of flash memory to store a "Person" and
// call it "my_flash_store".
FlashStorage(my_flash_store, Person);

// Note: the area of flash memory reserved lost every time
// the sketch is uploaded on the board.

void setup() {
  SERIAL_PORT_MONITOR.begin(9600);
  while (!SERIAL_PORT_MONITOR) { }

  // Create a "Person" variable and call it "owner"
  Person owner;

  // Read the content of "my_flash_store" into the "owner" variable
  owner = my_flash_store.read();

  // If this is the first run the "valid" value should be "false"...
  if (owner.valid == false) {

    // ...in this case we ask for user data.
    SERIAL_PORT_MONITOR.setTimeout(30000);
    SERIAL_PORT_MONITOR.println("Insert your name:");
    String name = SERIAL_PORT_MONITOR.readStringUntil('\n');
    SERIAL_PORT_MONITOR.println("Insert your surname:");
    String surname = SERIAL_PORT_MONITOR.readStringUntil('\n');

    // Fill the "owner" structure with the data entered by the user...
    name.toCharArray(owner.name, 100);
    surname.toCharArray(owner.surname, 100);
    // set "valid" to true, so the next time we know that we
    // have valid data inside
    owner.valid = true;

    // ...and finally save everything into "my_flash_store"
    my_flash_store.write(owner);

    // Print a confirmation of the data inserted.
    SERIAL_PORT_MONITOR.println();
    SERIAL_PORT_MONITOR.print("Your name: ");
    SERIAL_PORT_MONITOR.println(owner.name);
    SERIAL_PORT_MONITOR.print("and your surname: ");
    SERIAL_PORT_MONITOR.println(owner.surname);
    SERIAL_PORT_MONITOR.println("have been saved. Thank you!");

  } else {

    // Say hello to the returning user!
    SERIAL_PORT_MONITOR.println();
    SERIAL_PORT_MONITOR.print("Hi ");
    SERIAL_PORT_MONITOR.print(owner.name);
    SERIAL_PORT_MONITOR.print(" ");
    SERIAL_PORT_MONITOR.print(owner.surname);
    SERIAL_PORT_MONITOR.println(", nice to see you again :-)");

  }
} 

void loop() {
  // Do nothing...
}`

The problem is that i can't save any type of data, once I write my name/surname, when i tried to compile and upload another time, it doesn't save anything in the flash storage. Could someone help me? I don't understand what's happening. Another question is, what's the correct form of reseting the nano 33 iot? i try to push the button center but i can't get anything.

ChathuZ, 2023-03-02:
Did you figure it out?

sdetweil, 2023-04-16:
because the lib uses the flash storage, which is shared with the sketch, each time the sketch is uploaded the flash is wiped clean to load the sketch..
the comments say as much 
```
// Note: the area of flash memory reserved lost every time
// the sketch is uploaded on the board.
```