arduino/ArduinoCore-samd #662: User-hostile library names (SBU, SDU, SFU, SNU, SSU)

alranel, 2022-02-08:
This core contains some built-in libraries that implement loading a sketch from various local storage mechanisms:

* **SBU**: loading from MKRNB module internal flash
* **SDU**:  loading from SD card
* **SFU**: loading from SPI serial flash
* **SNU**: loading from NINA module internal flash
* **SSU**: loading from MKRGSM module internal flash

Over-the-Air updates are the most common use case for such libraries.

The issue is that since those libraries have examples, and they are installed by default, they are very prominent in the IDE examples menu but their names are quite obscure:
[image]

An option (A) could be to rename them with more descriptive names such as `LoadSketchFromSPISerialFlash` or `LoadSketchFromNINA` but this has two drawbacks:

* they would be still very prominent and confusing for a basic user, considering their niche use cases;
* the rename would break dependent libraries such as https://github.com/jandrassy/ArduinoOTA or https://github.com/arduino-libraries/ArduinoIoTCloud.

Another option (B) could be to move them to separate repositories that they don't get installed by default but this is suboptimal for code maintenance as they depend on bootloader code which is located in this repository (and it would break dependent libraries as well).

Another option (C) could be to move the examples somewhere else so that they are not displayed in the IDE menu.

Another option (D) could be to intervene on tooling, extending the `library.properties` metadata and/or changing the IDE. Examples of possible actions:

* group all these libraries under a single "Load sketch from storage" category;
* mark them as "Advanced";
* show the short description from `library.properties` so that understanding the purpose of a library is not entirely delegated to its name.

Comments?

alranel, 2022-02-08:
Comment from @per1234:

> My main concern is to make sure that if a user sees a bundled library in their menus and wonders what it is, it is easy for them to find the answer. From a quick look, it seems that the "Usage" examples do a pretty good job of explaining the libraries.
> The one that seems a little confusing is "SAMD_BootloaderUpdater". The example and metadata give me the impression this library is for any SAMD board:
> 
> ```
> This sketch checks if your Arduino SAMD board is running the latest bootloader.
> If it is not, it prompts you to update it.
> ```
> 
> But this is only for the MKR Vidor 4000 and compiling for any other board will give an "Unsupported board!" error.
> It is true that the example comment header says:
> 
> ```
>   Circuit:
>   - MKR Vidor 4000
> ```
> 
> but that might be read as an example of a board the library can be used with, rather than a comprehensive list of the supported boards.

Xtreemtec, 2022-06-16:
I agree the names are "difficult"   And just to open every  example to see what it does is not a good way.

There is plenty of room behind the acronym to decribe the lib.