arduino/ArduinoCore-samd #673: Support for nano_33_ble

tomers, 2022-06-12:
Hello there,

I hope this is not a stupid question.
I own nano_33_ble, and the closest variant supported by this project is nano_33_iot. I would like to ask whether I can build for the BLE using the IOT variant. If not, how should I add support for the new variant - should I just copy the `nano_33_iot` directory to `nano_33_ble` and make changes there, or is there other way around?
I am sorry for not being specific. I am not familiar with this project, so I have a lot of learning curve to do in order to find this answer on my own. I could use some guidance here, and hopefully I would be able to contribute to this project by adding support for yet another variant.

Thanks,
Tomer

per1234, 2022-06-12:
Hi @tomers. The [Arduino Nano 33 BLE](https://store.arduino.cc/products/arduino-nano-33-ble) uses a microcontroller (nRF52840) of a different architecture from the SAMD architecture of the board supported by this platform (ATSAMD21G18).

So it would not be possible, or even make sense to add support for the Nano 33 BLE to this project. There is already a separate boards platform that provides support for it here:

https://github.com/arduino/arduinocore-mbed

tomers, 2022-06-12:
Thanks @per1234 for the prompt response. I will look there

closed 2022-06-12 by tomers