arduino/ArduinoCore-samd #678: Serial Flush blocks forever but not only at programm start

Someaccound, 2022-08-31:
Hi there

I have a similar issue to #597, but not at the start of my program.

In my code, there is a function which looks like this:

```
void SlaveWaterCOM::send() {

    digitalWrite(2, HIGH);

    Serial1.write(outbuffer.dataarray, 28);
    
    Serial1.flush();

    digitalWrite(2, LOW);
}
```

The function communicates via the Serial1 over a MAX3075, and therefore it needs an enablepin for the driver.

In my program, this is the only point I use the Serial1.write or the Serial1.flush command.

Sadly, the program sometimes hangs at the Serial. Flush() command. 

I tracked the issue down to the Sercom->USART.INTFLAG.bit.TXC flag, which seems to never be written.

Has anyone a clue why this could be?

Duality4Y, 2024-12-30:
I have this exact same issue, it hangs on communication using a sercom module when calling flush() at some point,
I ALSO see with a debugger that the TXC bit is NOT set.
(JLink + gdb )

But In my specific case it seems to also be related to putting the SAMD to sleep, as when I disable this functionality it seems to run fine and stable.

It doesn't happen on the first wake-up and sleep cycle but somewhere down the road at some Time X
But that is in my case.

xsensefw, 2026-11-06:
We hit this in the field on a SAMD21 board (deep sleep between transmissions, modem on another UART) and tracked it down to the core. The cause is `SERCOM::acknowledgeUARTError()`:

```cpp
void SERCOM::acknowledgeUARTError()
{
  sercom->USART.INTFLAG.bit.ERROR = 1;
}
```

A bit-field write is a read-modify-write of the whole `INTFLAG` register. With arm-none-eabi-gcc 7.2.1 (`-Os`) it compiles to:

```
ldrb  r1, [r2, #24]   @ read INTFLAG
orrs  r3, r1          @ set ERROR (0x80)
strb  r3, [r2, #24]   @ write it back
```

`INTFLAG` flags are cleared by writing 1, so writing back a TXC that is already set clears it.

The sequence that hangs:

1. The sketch writes to the UART. `writeDataUART()` sets `onFlushWaitUartTXC = true`.
2. The last byte is shifted out and TXC is set.
3. Before `flush()` is called, a receive error happens on RX, e.g. a frame error from noise on a floating or weakly pulled-up RX line.
4. `Uart::IrqHandler()` calls `acknowledgeUARTError()`, which clears ERROR **and TXC**.
5. `flush()` → `flushUART()` waits for TXC while `onFlushWaitUartTXC` is still true. Nothing more is sent, so TXC is never set again, and it waits forever.

The window between the last write and `flush()` can be long (any code that runs in between), so a noisy RX line makes it likely. In our fleet it shows up only on a subset of units, which we believe pick up glitches on the unconnected RX line. On the hung units TXC = 0 with an empty TX buffer, as reported above.

Fix: write only the ERROR bit, `sercom->USART.INTFLAG.reg = SERCOM_USART_INTFLAG_ERROR;`. PR: #744.

Workaround until then: in the sketch's `SERCOMx_Handler()`, clear a pending error with `INTFLAG.reg = SERCOM_USART_INTFLAG_ERROR` (and the STATUS error bits) before calling `SerialX.IrqHandler()`, or avoid `flush()` and wait for `availableForWrite()` to reach the buffer size with a timeout.