arduino/ArduinoCore-samd #679: Udp.h included in ArduinoCore-samd and in ArduinoCore-API

PatrickDekker98, 2022-08-31:
Reference from ArduinoCore-API repo: https://github.com/arduino/ArduinoCore-API/issues/171

Udp.h is included in Core-samd and core-api, while one of them uses #pragma once and the other include guards. This causes problems when including both of them. 

`error: redefinition of 'class arduino::UDP' `

matthijskooijman, 2022-08-31:
As indicated in the other issue, I think that Udp.h should probably just be removed from this repo.

@PatrickDekker98  Did you happen to have checked the contents of these files? Any functional differences (i.e. other than comments and spacing)?

PatrickDekker98, 2022-08-31:
Both classes are pure virtual and practically do the same thing: nothing
But, both classes have the same contents too

PatrickDekker98, 2022-08-31:
@matthijskooijman do you know why the arduinoCore-api is not a git submodule in this project? 
To me development of arduinoCore-samd seems cumbersome, having to first download api, then samd, then creating a symlink from api to samd

matthijskooijman, 2022-08-31:
See https://github.com/arduino/ArduinoCore-API/issues/96 for some consideration about using sumbodules.