arduino/ArduinoCore-samd #700: Serial buffer size unchangeable

maxgerhardt, 2023-04-02:
Per https://github.com/arduino/ArduinoCore-samd/pull/651#issuecomment-1492773652.

Due to the `#undef` construct done in

https://github.com/arduino/ArduinoCore-samd/blob/a5d52dbf7bcf338de2042eb74cfba33993729c81/cores/arduino/Uart.h#L25-L29

it's not possible to set a global `-DSERIAL_BUFFER_SIZE=...` value and have it be use used anymore.

A user has a usecase for that [here](https://community.platformio.org/t/serial-buffer-size-on-grand-central-m4/32991).

Muny, 2023-07-21:
Agreed, it was pretty deleterious to add those.

maxgerhardt, 2023-07-21:
Still seems like the amount of people that care about this is exactly 0.

Muny, 2023-07-21:
> Still seems like the amount of people that care about this is exactly 0.

I presume it's because the people who do care about it find ways to work around it. I ended up adding a pre-build step to my project to patch this file.