arduino/ArduinoCore-samd #711: Alternate SERCOM I2C Not Working On Pins PA08 and PA09 (4 and 3)

DeltaAerospace, 2023-10-19:
I2C is not working using PA08 and PA09 on the ATSAMD21G18A-AU with an Arduino Zero bootloader on a custom PCB.
-Tried Modified I2C Scanner
-Normal I2C on PA22 and PA23 still work

wltue, 2024-05-21:
Works for me. Problems could be init in the library I had a lot of issues with external libraries if != Wire'0',
If the library tries to init the Wire remove the line (.begin) and add it manually in your setup.

Try manualy, this works:

`#include <arduino.h>
#include "wiring_private.h"  // for pinPeripheral() function
#include <Wire.h>

TwoWire Wire1(&sercom2,  // &PERIPH_WIRE
              4,         // PIN_WIRE_SDA
              3);        // PIN_WIRE_SCL

void setup() {

  // Init Wire1
  Wire1.begin();
  Wire1.setClock(100000);
  pinPeripheral(4, PIO_SERCOM_ALT);
  pinPeripheral(3, PIO_SERCOM_ALT);

  Serial1.begin(115200);
}

void loop() {

  Wire1.beginTransmission(0x40);

  Wire1.write(0b101010);         // Sends value byte
  Wire1.endTransmission();  // Stop transmitting

  delay(500);
}
`
(set the parameters according to your i2c chip connected)

DeltaAerospace, 2024-05-26:
Since I posted this, I’ve got it working. Thanks for the help.