arduino/ArduinoCore-samd #726: cores/arduino/api directory is missing

abetis, 2025-02-12:
Hi.

I'm trying to create an environment for my custom board based on the SAMD21 chip.

The code is based on the latest 1.8.14 version of the [ArduinoCore-samd](https://github.com/arduino/ArduinoCore-samd)

The [Arduino.h](https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/Arduino.h) includes the "api/ArduinoAPI.h", while the "api" directory doesn't exist in the repository...

I've checked the packages installed by the IDE on my computer "packages/arduino/hardware/samd/1.8.14/cores/arduino/" and there is an "api" directory there...

Am I missing something? Shouldn't the "api" directory be in the repository as well?

Thanks.

pennam, 2025-02-12:
You should find all needed information here
https://github.com/arduino/ArduinoCore-samd?tab=readme-ov-file#developing

abetis, 2025-02-12:
Oh. Didn't expect that there will be 2 repos that need to link. 
Thanks.

closed 2025-02-12 by abetis