arduino/ArduinoCore-samd #729: OTA functionality example does not work

Gaveroid, 2025-03-20:
So I've been pulling my hair out for days trying to figure out how to do OTA flashing on my MKR WiFi 1010 with a locally compiled binary, served over HTTP from S3 or otherwise my own web server. I finally stumbled across the Usage.ino example under the SNU library in this repository.

[libraries](https://github.com/arduino/ArduinoCore-samd/tree/master/libraries)/[SNU](https://github.com/arduino/ArduinoCore-samd/tree/master/libraries/SNU)/[examples](https://github.com/arduino/ArduinoCore-samd/tree/master/libraries/SNU/examples)/[Usage](https://github.com/arduino/ArduinoCore-samd/tree/master/libraries/SNU/examples/Usage)/Usage.ino

This example itself does not work for me. The arduino_secrets.h file as-is links to a binary that I confirmed does exist and is available for download, looks like some kind of RGB blink example sketch, yet it will not update to that sketch. The only modifications I made to the file were inputting my wireless credentials.

See here, what I observe in my serial monitor:

`Attempting to connect to SSID: (my actual network SSID)`
`Sketch compile date and time: Mar 20 2025 16:44:43`
`Type "download" in the Serial Monitor to start downloading the update`
**I typed download**
`Downloading update file`
`Download complete, please restart or type "restart" to apply the update`
**I typed restart**
`Filesize: 167`
**device restarts right back into Usage.ino**
`Attempting to connect to SSID: (my actual network SSID)`
`Sketch compile date and time: Mar 20 2025 16:44:43`
`Type "download" in the Serial Monitor to start downloading the update`

I am on WiFi NINA firmware 2.0.0.

I'm not sure if there's something wrong with my board (be it the bootloader, WiFiNINA firmware, etc.) or what, but at this point I feel I have ruled out an error between keyboard and chair because other than my credentials, unmodified example code is not performing as expected. I'm open to any ideas though, I'd be perfectly happy to hear if it is something I'm doing wrong as long as I get it sorted.

pennam, 2025-03-21:
Hi @Gaveroid there might be something wrong with the file download. The reported filesize of 167 is not the real size of the binary. If you do a `wget http://downloads.arduino.cc/misc/WiFi1010_blinkRBG.bin` you will get a file of 30528 bytes.

I guess plain http is disabled from `download.arduino.cc` so I've tryed to switch to https but got an authentication error.

I'll keep you posted

pennam, 2025-03-21:
@Gaveroid i confirm the issue is http download is no more possible from `downloads.arduino.cc` if you download the file with `wget` and create a local http server you will be able to download the file and make an OTA.

I've opened a PR in the nina-fw repo to enable https download https://github.com/arduino/nina-fw/pull/112 I'll ping you as soon as a new firmware version is available

Gaveroid, 2025-03-21:
@pennam Wow, I'm so surprised I didn't catch that low file size. I was paying attention to file sizes and ensuring they matched when trying to load my own binary, but guess I got sloppy when trying the example .ino. I also knew about https not being supported, but hadn't considered that maybe `downloads.arduino.cc` disabled http since that code was written. Thanks for figuring that out and getting back to me so soon!

I was able to successfully spin up a local http-only web server and flash my board OTA with `WiFi1010_blinkRBG.bin`. It does execute as intended and after restarting, I see the on-board RGB LED cycling through colors. However, I am unable to get a binary I compiled myself to load and I'm not sure why.

I have a very simple serial test sketch, seen below. While my true intention is to flash a much larger firmware, and I am sure there is some file size limitation on what I can send OTA, I am trying with something very simple to start.

```
void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println("Test!");
  delay(1000);
}
```

It does operate as expected when uploading directly to the board over USB. For OTA use, I'm generating the .bin file by going to `Sketch > Export Compiled Binary`. Then in my sketch's directory, under `/build/arduino.samdmkrwifi1010` I see a variety of files as listed below.

```
SerialTest.ino.bin
SerialTest.ino.elf
SerialTest.ino.hex
SerialTest.ino.map
SerialTest.ino.with_bootloader.bin
SerialTest.ino.with_bootloader.hex
```

I am under the impression that I am meant to use ``SerialTest.ino.bin``, the normal binary with no bootloader included. I can confirm the file is being accessed over http & downloaded through the same local web server that I used to host ``WiFi1010_blinkRBG.bin``. Except when I go to restart after downloading, the board is back running ``Usage.ino`` trying to download it all over again. My own ``SerialTest`` binary does not get applied.

I tried it "as nature intended" first, but I am pretty sure there is some necessary post-processing of the binary. I have tried running it through ``ltss.py`` to compress it, because I read that ltss compression is expected. No dice. I also ran it through ``bin2ota.py`` under the impression that there is some header / length / etc standardization that is expected. No dice. Neither my ``SerialTest.ino.bin``, ``serialtest.py`` or ``serialtest.ota`` will work.

Hopefully there's some simple step I'm missing. I greatly appreciate all your help so far and am so happy that I've been able to successfully OTA flash the board with ``WiFi1010_blinkRBG.bin``, now I just have to figure out where I'm going wrong trying to prepare my own binaries for OTA flashing.

Gaveroid, 2025-03-21:
Update: I have also tried directly running ``SerialTest.ino.bin`` through ``bin2ota.py

Gaveroid, 2025-03-21:
Update: I also tried running ``SerialTest.ino.bin`` through ``bin2ota.py`` directly without running it through ``lzss.py`` first, with no luck. I was under the impression that lzss compression was to be done before bin2ota, but I decided to just try bin2ota straight anyway, again no luck.

I also accidentally managed to close this by accident but I have reopened. Whoops.

pennam, 2025-03-21:
@Gaveroid `lzss.py` and `bin2ota.py` are needed only if you want to make a binary in the Arduino Cloud OTA style.

I think you are missing to 

`#include <SNU.h>`

in your update sketch.

Gaveroid, 2025-03-21:
> [@Gaveroid](https://github.com/Gaveroid) `lzss.py` and `bin2ota.py` are needed only if you want to make a binary in the Arduino Cloud OTA style.
> 
> I think you are missing to
> 
> `#include <SNU.h>`
> 
> in your update sketch.

This solved my issue. I thought ``SNU.h`` only had to be imported into the firmware I was using to handle the OTA download, not the target firmware. Once I included it in my test program, I was able to successfully update OTA. 

Of course in my actual use case I will be updating to a new variation of my current program, so firmwares on both side of the update will have OTA handling functionality. I knew I wouldn't be able to OTA update again without flashing back to the first firmware, I was just trying to write a really simple sketch for testing before I integrated it and didn't expect to need it in there too.

I do look forward to a version of ``nina-fw`` with HTTPS support, but this is perfectly usable for now. Thanks so much for your help!

closed 2025-03-21 by Gaveroid

JAndrassy, 2025-03-21:
If you want to upload the update directly from IDE, without a http server, you can use my ArduinoOTA library. it is in the Library Manager

Gaveroid, 2025-03-21:
> If you want to upload the update directly from IDE, without a http server, you can use my ArduinoOTA library. it is in the Library Manager

I appreciate the heads up! Great to know that this is an option for future use. However, it seems to me that your library targets local network use and for my current application I intend to update over the internet.