arduino/ArduinoCore-samd #730: Raspberry Pi install fails with "Tool avrdude not available for your operating system"

robrennie, 2025-03-24:
When trying to install versions 1.8.14 or 1.8.13 of "Arduino SAMD Boards (32-bits ARM Cortex-M0+)" via the Arduino IDE 1.8.19 Boards Manager on a Raspberry Pi, the error message _"Tool avrdude is not available for your operating system" and fails._

**Specifics:**

- Raspberry Pi 5, default image.
- Arduino IDE 1.8.19 - the only version available supported for my processor (arm64).
- This was working fine on a different Raspberry Pi for well over a year. Then this morning "something" changed - not sure what, and I started getting that error. Then, I tried a fresh new Raspberry Pi (5) and am now seeing the error there too.

**I've done some research:**

- It appears the SAMD Board package (index) has a requirement on avrdude 6.3.0. I am using the MKR Wifi 1010 board.
- The one Raspberry Pi that still works has avrdude 6.3.0 installed.
- The two non-working Raspberry Pi's somehow have avrdude 7.1 installed. Thus, the error message appears correct, it can't find avrdude 6.3.0.
- I tried manually changing the package index to version 7.1, but the Arduino IDE just recognizes the change and overwrites it with the original 6.3.0.
- I tried finding avrdude 6.3.0 to re-install, but it's not available anywhere.

kilitary, 2025-04-11:
give points to your projects.
select local only development, when cautching errors or other miss-yesterday errrrrrors.

controllercustom, 2026-09-23:
I removed a tool dependency on avrdude to get around this problem.
https://forum.arduino.cc/t/arduino-samd-core-2-versions-behind-on-arm-hosts/1460569